x-station/xstation-issues #326: PU-18 Stopped working and boots to CD audio or main menu

guillemdc, 2023-11-04:
Hi,

After some time not using my PS1, I came back to play again and I got the CD menu with the lid closed or the main menu with it open.

I hadn't touched the SD-Card, however I've read it with my PC and it seems everything is in order.

I've probed looking for continuity using [this guide](https://user-images.githubusercontent.com/5185337/100875797-6641d680-349e-11eb-85f8-eb8d9cd4f155.jpg) and everything checks out.

However, I realized these three pins (see attached pic) are connected together to ground and I'm not sure that's OK. **Could you please confirm?** Even though the solder joints look OK and the thing used to work fine, maybe something was too close to ground and with time and temperature changes it has suddenly touched it causing the issue. I don't know. I'm honestly running out of ideas.

[image]

Regards

guillemdc, 2023-11-04:
Nevermind. I put the software on a new sdcard, popped it in and worked like a charm. Who know why the other one doesn't work, but I can play again and that's what matters. Sorry, I should have checked that before posting.

closed 2023-11-04 by guillemdc

ramapcsx2, 2023-11-04:
Good good :)
The 3 points marked should all be ground, by the way. That's correct.