x-station/xstation-issues #327: gameshark?

pcaudill3323, 2023-11-14:
i wonder if you can put gameshark in the xstation, it would be great to have and have it remember the codes for the games as well, so you don't have put them in every time you play a game.

ramapcsx2, 2023-11-14:
You can get a Gameshark CD version that starts before your game. For this, create a game queue, selecting the GS disk first, your game second.
Start the queue, select your cheats, and then open/close the CD lid to swap to your real game.

Of course you can also simply install a cheat cart as usual. Those will work entirely normally :)

pcaudill3323, 2023-11-18:
it would be great if you made a video showing how.

ramapcsx2, 2023-11-18:
Oh, I'm sorry but I'm always working as it is. I'm not any good at making videos, nor writing guides :p 
I can help you understand, but please don't expect like.. a video :p 

Disk queue system:
- Highlight your first CD, in this case a Gameshark CD image or similar, and press Square
- Highlight your game and press Square
- You now have 2 games queued up. Press Start to run it.
- Open/Close lid to swap between the 2 CD images when needed

I think the Carts are self-explanatory?
[image]

kaitlin459, 2024-04-16:
> it would be great if you made a video showing how.

i was able to get the cheat engine iso working no problems it even works with the memcardpro so no need for a gameshark just use the cheat engine compilation and your all set