x-station/xstation-issues #329: X station sound issues - FMV and Gameplay

JTAnimator, 2023-12-06:
I’m having problems with sound on many games when using X station. While games run fine like Star Wars Jedi Power Battles and Tempest X3, select sounds such as FMV intros and cutscenes are mute. During gameplay, you don’t music and sound effects.

My PS1 console model is 5502, I purchased it from a professional modder and the ROMS have been ripped from my own discs. I've read around saying that it could be an SPU problem, is there going to be a fix in future firmware or is there any way it can be fixed without having to mess around with it?

Thank you.

ramapcsx2, 2023-12-06:
Hey,
if you have a 2nd SD card around to test, could you put one of your problem games on it and try again?

JTAnimator, 2023-12-06:
I’ve put on games like Tempest X3, Spyro 2 and Star Wars Jedi Power Battles. I use 3 other Sandisk SD cards including a 128gb Sandisk Ultra. I still get the same problem.

Sound Effects play fine on Tempest X3, Spyro and Star Wars Power Battles, but in game music and FMV doesn’t play. In Spyro 2, the cutscenes play fine.

I also tried Twisted Metal World Tour and Moto Racer World Tour, both don’t play sound on FMVs but the rest of the games sound works fine.

ramapcsx2, 2023-12-06:
Okay, I think your SPU is indeed bad .. :/

ramapcsx2, 2023-12-06:
I've had some (1002) hardware in the past with exactly this behaviour, and I wasn't able to fix it back then. Missing audio from FMV sounds exactly like it.

ramapcsx2, 2023-12-06:
Additional info:
We know the xStation is generally fine, and works generally as expected with the PSX.
Games load, and FMV play. Those FMV then have a detail problem, where there's no audio in them
Audio and Video are generic CD data, so it's impossible to be a CD problem, since video works.
(This is how I'm sure it's no xStation issue.)
It could however happen if the SPU chip has some kind of fault, where it doesn't play XA audio it receives.
This is some kind of chip damage problem, and usually one has to swap the problematic chip to fix it (or swap entire board).

MathMarvel, 2024-08-14:
I had this same problem recently with a SCPH-1001 model PS1. It turned out that the chip responsible for decoding XA audio, CXD1815Q, was faulty. Replacing that chip resolved the problem. It does appear that the same chip is found on the SCPH-550x models.