x-station/xstation-issues #330: Action Replay Pro and Explorer FX carts won't load

a7mag3ddon, 2023-12-14:
Hi, not sure if this is an issue or if by design, I have a PAL 5502 PS1 and my AR2 Pro and Explorer FX carts won't boot.

I've cleaned both the carts and the parallel port on the ps1 but just won't boot up.

I tried the cheat carts on a pal 7502 and they load up perfectly every time.

Before I take the machine apart and start reflowing and cleaning the parallel port I just wondered if it is supposed to work or not ?

The xStation otherwise works perfectly on the system both in pal and NTSC modes playing games etc.

Thanks for any help or explanations forthcoming.

ramapcsx2, 2023-12-14:
Yea, the xStation does not affect the expansion port at all. I'd assume poor contact on the connector, which is quite common.
When I need to get a cart working on such consoles, it's usually a combination of thorough cleaning of cart + port that fixes it.
The port is tough to free of the oxide, but you can have some deoxit (or similar) on the cart, and then repeatedly attach / detach the cart. The hope is to clear away at least some of the oxide that way.

a7mag3ddon, 2023-12-14:
Thanks for confirming now I know to concentrate on the connecting

a7mag3ddon, 2023-12-14:
Resolved

closed 2023-12-17 by ramapcsx2