x-station/xstation-issues #331: Video CD files with Gamars Movie Card possible?

badass-boss, 2023-12-19:
So I just got this thing called Gamars Movie Card PSX-003 that lets you play video CD discs on your PS1. It has a switch for VCD or password. When I boot it up in VCD mode, it waits and give me a prompt to insert a VCD which I obviously cannot do. When I boot it up in password mode, it brings up its own cheat menu and when I press start, I can finally see Xstation menu. When I choose a game, it brings up that cheat menu again but this time when pressing start, the game starts up so it's sort of a cycle of menus here

I tried to circumvent it and when I get to Xstation, I choose a video CD file and once it's in the cheat menu again, I flip the switch to VCD and sure enough it starts loading the VCD file! Or does it?? Well......... some VCD files didn't work and it complained the disc is probably corrupted. Other movies booted up but it skipped frames, sometimes severely and other times movies had no audio and other times movies just crashed while playing. Is there anything wrong with my VCD files? I must say they play super janky but they are consistent every time. They always glitch the exact same way on the same moments. Is there a way to improve this? Granted I haven't tested this thing yet with another real PS1 with a real CD drive. Also does anyone know of any decent VCD creating software that is not 2000s sh*tware full of viruses?? Any ideas will be appreciated

ramapcsx2, 2023-12-25:
VCD support should definitely be doable. The other day, I got this working on the (wip) mStation, so there's little reason it can't also work on xStation.
We'll see :)

ramapcsx2, 2024-06-23:
Soon :)