x-station/xstation-issues #332: Delete games from  Xstation Menu

Dert2k, 2024-01-08:
Any way to delete games? As i go through games i would like to delete the games i know i wont play.
Feature request maybe?

ramapcsx2, 2024-01-09:
Hey, I thought about this before.
In trying to avoid SD writes and possible card fragmentation, I would implement this as simply hiding games from view.
It's a good feature to have, and already on a list somewhere :)

Dert2k, 2024-01-09:
yes fragmentation would occur which we could use a computer to defragment or moving the games out of the SD card and moving them back in. 

If deleting is not possible, 
Another good add on could be allowing us to add a check ✅ or an X ❌ next to the game titles. This way would could mark games and keep a list and an option for “Show only Checked” or “show only X”

and thinking of this another feature could be “played” and “not played”

most played… etc