x-station/xstation-issues #333: xstation power led blinking blue?

lf94, 2024-01-18:
Ive tested all the joints and they all seem fine... any ideas? thanks!

[image]

lf94, 2024-01-18:
Well, I'm seeing now that Stone Age Gamer has not only 1. Forgot the QSB in the first order but 2. Has sent me **a broken board**.

This is what the board should look like: 
[image]

Very, very frustrating.

I will be manually wiring the missing traces and sending an email to SAG just to get another out of principle.

lf94, 2024-01-18:
@ramapcsx2 what's the Discord link by the way? Every one I've found is dead :D Thank you! :heart:

lf94, 2024-01-19:
To anyone coming here that was my exact problem. It works now!

[image]

closed 2024-01-19 by lf94

ramapcsx2, 2024-01-19:
Hey, sorry I spotted this late :p 

SAG will send you another spare, as you say, out of principle. These QSBs have a weak point in shipping, with that hook breaking quite easily. We have a redesign in the works.

Your repair should consist of replacing the missing system ground connection, which I think you have done here.
The other detail is also right: You restored grounding the PsNee gate connection.
All in all, this looks good. One point that you should improve though: Please shorten that APLL wire :)