x-station/xstation-issues #334: Bouncing video on very old CRTs?

lf94, 2024-01-19:
So my video output is essentially perfect with my new install on a 5501! The only thing that's absolutely odd is a bouncing. For what it's worth, this is a very old Zenith CRT TV (one of those ones in a wooden container, weighs a massive ton). I'm thinking, is it possible the XStation isn't generating an as-perfect signal as the original one in the Playstation?

Thank you very much!

ramapcsx2, 2024-01-19:
Wow, that is obscure, but old Zenith TV have had a Sony original hackfix back in the day. If you had a Playstation and this problem, Sony would install a weird sync modifying circuit in your console.
These can still be found occasionally in the wild, but I have no idea what exactly they do.

It's a problem with the console directly though, not particularly related to the xStation :p

lf94, 2024-01-19:
Haha, nice to know! I guess my PS1 will be for my more modern tiny CRT I've got ;)

It would be dope if we could reverse engineer that circuit or whatever too, but I understand you're probably doing all this as a side hustle and wouldn't want to mess around :) I'm not sure how I would even measure things. I have a logic analyzer but it's a very cheap one with low sample rate.

lf94, 2024-01-19:
https://www.reddit.com/r/retrogaming/comments/oyzl7q/playstation_1_and_zenith_tv_bouncing_image/

lf94, 2024-01-19:
Aha! Extremely nice pictures: https://imgur.com/gallery/bX1YLSE

lf94, 2024-01-19:
If anyone tells me the parts off mouser to get and sends me a PCB, I will 100% try this out.

ramapcsx2, 2024-01-20:
That's the thing, yep!
Somehow particularly an old Zenith can't deal with the otherwise pretty tame PSX syncs.
I wonder whether other old consoles work on it? :p 

You could try to reverse engineer what this circuit does.

lf94, 2024-01-20:
I guess surprisingly all my consoles, from NES to N64, and then HDMI<->Composite have all worked :) Even PS2.

Yeah I may reverse it. It seems that one flip-flop chip does all the lifting on there, which is easy to source from mouser...

lf94, 2024-04-13:
How come you re-opened? May we close this?

ramapcsx2, 2024-04-15:
No idea, sure :p

closed 2024-04-15 by ramapcsx2