x-station/xstation-issues #335: No start up sounds or sound effects in game (bgm is fine)?

deelowe, 2024-01-21:
I'm having an odd issue with my xstation install. There is no start up sounds and when I play a game, only background music is played, no sound effects. I think my installation is pretty good. I checked everything over with a loop and a multimeter.

Can someone who is more familiar with the PS1 architecture point me in the right direction to troubleshoot? I have oscope, good multimeters, soldering station, etc.  Just not sure where to start. 

Thanks.

[image]

lf94, 2024-01-21:
It'd be good to get a pic of the lifted pins too

deelowe, 2024-01-21:
Lifted pins:

[image]

[image]

ramapcsx2, 2024-01-21:
The issue as you describe it is clearly related to the SPU (CXD2925) or its RAM chip.
Using your scope, you could try to probe all the signals around the SPU when it runs, looking for any clearly misshaped digital signals, which I often find in these situations.
Example of exactly that: https://imgur.com/09S9Tts

deelowe, 2024-01-22:
@ramapcsx2 thanks. Is the issue unrelated to the xstation? Trying to isolate things. The PS1 has never been recapped and has definitely seen some wear so I wouldn't be surprised if something has gone bad.

ramapcsx2, 2024-01-22:
Yep, it's chip damage on the SPU. The xStation doesn't touch it :p

deelowe, 2024-01-23:
Yep. Chip appears bad. Thanks for the help. Closing this issue as it's not related to the xstation.

closed 2024-01-23 by deelowe

ramapcsx2, 2024-01-23:
I want to add that it's sadly at elevated levels with PU-8 boards.
Whenever I have a random PU-18, I'm sure that it'll work. With PU-8, I just "hope" :p