x-station/xstation-issues #336: General SD Card Question

pirsir76, 2024-01-22:
Hello,

I don't own an Xstation yet but I'm researching.
What is the max size SD card supported?
Can the xStation be used with an SD adapter so it can be used with micro SD?

And, I don't suppose it supports controller reset?
Like, you press a button combination on the game pad and it will reset the system?

Is there a compatibility list for PS1 titles?

Thanks :)

lf94, 2024-01-22:
Not the dev but I can say that it uses micro SD by default.

I don't know the max size it can access, but presuming because it supports exFAT, something extremely large.

---

Pretty sure there is no controller reset feature. I see people have additional mods for this.

I think the compatibility list is extremely good given this system has one goal: simulate a CD drive. With the latest firmware, FMV and whatnot play very well.

ramapcsx2, 2024-01-23:
Hey ho, let's see..
The default storage format is microSD, with sizes up to 1TB tested, and this is likely to be the maximum.
The xStation is a low level ODE, which means it has very limited interaction with anything but the CD system.
It means that compatibility is very high, but features such as the IGR can't be done by it. Others have stepped in with mods for this.
With regards to a compatibility list, well, I simply strive for 100%. Occasionally a title slips through, but it'll be fixed in an update.
Expect everything to work :)

pirsir76, 2024-01-24:
Hey thanks for the response ramapcx2.
Always good to speak to the dev directly.
I'll probably be getting the xStation and professionally installed, so there wont be any issues.
I might even test a 1.5TB microSD or 2TB (when they are released) always good to have that extra storage.

Thanks, I'll close this issue now.

closed 2024-01-24 by pirsir76