x-station/xstation-issues #339: Xstation not loading into loader menu

lukis142, 2024-01-26:
Hi, I have a freshly installed Xstation in a SCPH-5502 system that just doesn't want to work. It can't seem to load the menu on startup, the LED jsut flashes twice very quickly and then nothing. However, I can go into the CD player and play Track 1 and it seems to be loading from the Xstation. This PS1 also had this problem on the original drive but I hoped that the xstation would fix this problem. All the solder joints look good and were tested with a multimeter. Has anyone ever encountered this issue?

ramapcsx2, 2024-01-27:
Oki, so you did the CD player test, and it shows the seconds advancing normally. This means that most signals are correctly working, but there is a problem with the console decoding any data.
It could be a damaged Decoder chip, though this is rare on PU-18.
Instead, I would check the 3 IS2 signals again, measuring resistance towards ground and each other.
[image]
The I2S signals are on 4,14,15, shown in the pic. Especially the crowded area around 4, 14 can have shorts towards ground, or shorts of the signals toward each other.

lukis142, 2024-01-27:
All of them show some resistance between eachother and ground (around 83-90kohms). Which one is the decoder chip? I have a feeling it might be bad, asi this PS1 refused to load games even from working CD drives

ramapcsx2, 2024-01-27:
1815Q on PU-18 is the decoder. If this is bad, the system can't decode the data, even if xStation sends it fine.

~80kOhm towards everything is good. If any of these were a really low value, something around 1000 Ohm, it would be fishy.. but your values sound good.

sicounet, 2024-01-27:
I had it with a pu-18 a few month ago , i have it again today . I did the cddaloader test where i should here a music with track 1. I have the 2:32 track but i hear no sound this time .

sicounet, 2024-01-27:
On the 1st pu-18 i had sound with the 2:32 track... i changed the 1815Q . That changed nothing .

sicounet, 2024-01-27:
This issue is so weird. It happpened to me twice with pu-8 and thought when it happened with a pu-18 that it was a rare issue ...but now i start to have doubts.

lukis142, 2024-01-27:
What's weird is that my track only has 30 seconds, not 2:32 minutes like you said

sicounet, 2024-01-27:
maybe it was a different test , i was in the same position as you and somebody told me to do this test with a file he gave me . You can find it in a similar issue closed.

ramapcsx2, 2024-01-27:
The track is the data for the loader, which is usually /00xstation/loader.bin.
If you place a CDDA dump there and name it "loader.bin", you should be able to hear the music. This works even if the decoder is bad.
Now, this loader has changed sizes at a few times with firmware updates, and the latest 2.0.2 firmware has one embedded even, so it works without any SD card at all. This is why your track times are different.

@sicounet if your track times make no sense, or the time counter is advancing weirdly, the problem is related to another set of signals (SUBQ related), not the clearly data decoding fail that @lukis142 has.

sicounet, 2024-01-27:
It makes sense and the time counter advance normaly . It's what the guy gave me in the issue i opened few months earlier . I use this test everytime i have suspected issue with the decoder . But for this time , the track advance normaly but i hear no music . That's a first .

ramapcsx2, 2024-01-29:
@sicounet Okay, but as I said, the CD player test can only check for SUBQ and support signals, not the I2S data decoding that the Decoder has to do. Testing the Decoder is the one thing it can't do :p

lukis142, 2024-02-19:
I have replaced the CXD1815Q decoder chip and now the console boots fine to the loader and it loads games just fine.

ramapcsx2, 2024-02-20:
Okay, it's something that afflicts some PU-18 as well. It would be great to come up with a test for this, but atm I have no idea.
Enjoy! :)