x-station/xstation-issues #340: Long press Reset doesn't reset to menu - Only restarts the Game - Late PU-8

thesunnymachine, 2024-01-27:
I tested all the points, cleaned the reset button, installed an in game reset as well.  The xstation is only see a short press on the reset switch. 

Holding the reset button causes the Green LED to blink rapidly.   Reset works perfectly on a pu-18 board with an xreset installed. No issues.  Same ribbon, board, SD card etc.. 

I checked all the solder points as well, they are making good contact.  Any ideas?

ramapcsx2, 2024-01-27:
Hmm, so this happens when the xStation doesn't see the reset signal, or sees it with interruptions etc.
If you have a different PSU around to test, it would be good to know whether a PSU swap fixes it. (It'd mean the fault is on the PSU, otherwise it means the fault is more upstream.)

thesunnymachine, 2024-01-27:
I only have later pu -18 models and the power supplies don't fit.

I'm going to recap the board and supply next week anyway.  I will also do some testing on the reset switch to see if it's maybe dropping signal when held.  The weird thing is that the long press does seem to reset the entire system as it acts like it's going to load the xstation, but then takes a bit to load the game. 

Which point on the pu-8 breakout board is for reset? I'll try and fix up the solder on that one point and check continuity to the switch signal, if possible?

ramapcsx2, 2024-01-27:
Signal "17" in this picture is the global Reset line.
When you watch it on a scope (or multimeter in a pinch), it should normally read 3.5V'ish and 0V when Reset is pressed. The low time should be regulated to some 200ms or more by the PSU.

https://github.com/x-station/xstation-issues#late-pu-8--21-and-later

thesunnymachine, 2024-01-27:
> Signal "17" in this picture is the global Reset line. When you watch it on a scope (or multimeter in a pinch), it should normally read 3.5V'ish and 0V when Reset is pressed. The low time should be [...]

Okay, here's what I have found, so far.  Voltage seems to behave as you describe.  I recorded a video of that here: https://www.youtube.com/watch?v=jr_WO7_FwPg

I removed the in game reset from the main board and disconnected the controller ports, no change. 

I thought maybe the issue was that the board firmware was setup for the previous pu-18 board so I deleted the files off the SD card and put fresh firmware files on the card.  I then reinstalled the firmware as well.  The information screen showed the console should reboot on it's own but that never happened.  I waited about 4 mins on this screen:
[image]

Here a video of me waiting and manually rebooting: https://www.youtube.com/watch?v=8BjKcDhROsA

So it looks like maybe the xstation is having trouble with the reset functions overall on this console, even after the firmware updates.  

I reflowed all the solder points on the quick solder board and I even reflowed the pins on that chip that the reset point is connected to, no changes.

Do you still think it's related to the power supply?  I can order a replacement to test, or if you have any other suggestions on what I can try, let me know.

thesunnymachine, 2024-01-27:
Good news!  I have found the issue.  After inspecting the quick solder board, it looks like the connector was not preset on the board to the trace.  So I ran a wire from the point on the board to the pin on the connector.  All is working now.  Somehow while doing all this testing I also blew one of those 1210 72v 2a Fuses as well, because suddenly the xstation had no power.  Luckily I was able to quickly find the blown fuse and replace it. 
[image]

I will try and replace this QSB with a new one when I order another xstation in the future... or maybe I will just leave it, since it's working. :)

ramapcsx2, 2024-01-29:
Yea that firmware update was a good way to test it. It confirmed there was no connection at all :p