x-station/xstation-issues #342: Nightmare Creatures boot problem

Z3R0XZ, 2024-02-05:
Nightmare Creatures sometimes it takes like 5 tries to boot. The game has 7 executables, and the first one is just a video player to play a STR video, I don't konw if that helps.

Z3R0XZ, 2024-02-06:
Oh, thanks for letting me know that is not an xStation problem. I might do a patch since I made a custom trainer for it. The code for the game is A$$ LOL.

ramapcsx2, 2024-02-06:
Oh awesome, I could <really> use some help from a PSX hacker for this! :D

So yea, what I reasoned is that the problem with this game is the repeated executable reloads at boot, which all use the super buggy kernel code. 
I suspect the game dies around this particular bug:
https://github.com/grumpycoders/pcsx-redux/blob/fce881226e30e11cca1370751932e8f36a67c59f/src/mips/openbios/cdrom/helpers.c#L37

It makes sense that reducing these reloads would then improve boot reliability :)

What would be awesome now though is, if you could look for further details on what exactly goes wrong with the game..
I'm not sure what can be done, but if the game had printfs around the reloads, maybe with some status results, that would help..
Another thing would be patching particular details, such as any delay loops (longer/shorter/remove) etc, anything really that gets the game to show some repeatable behaviour.
Right now, it is just all totally random, though I noticed the disk version dies less often than xStation. I have the PAL release here, but PAL/NTSC-U both do it.