x-station/xstation-issues #344: After installing xstation partial problem with analog modem on the driver.

PSXmanCZ, 2024-02-25:
Hello

I installed xStation botting games works everything is great until I want to connect to run analog mode on the original controller. And that's in the Bios and in the game Ape Escape.

xstation is on the latest FW
I have the PAL version of the SCPH-5502 console (motherboard 1-664-537-52)

I plug in the original driver, when I leave the lid open and go into the bios after turning on analog mode the control arrow disappears and the driver stops working.  In the xstation menu the controller responds with analogue on and in the dozen games I've tried the analogue stick works too. But when I start Ape Escape (PAL version) the intro can be skipped by pressing the button but then in the menu the controller doesn't respond. But when the demo shows up after a minute of not responding I can skip it but then nothing in the menu again it's weird.  The only game that requires analogue and then it's a problem.

I have checked the fuses on the board and all the zener diodes are leaking and the whole 8V circuit to see if there is something wrong. I checked all the points of the QSB to see if it is connected to the motherboard correctly also if all the points lead to the tape socket and if the QSB itself is defective. And that was fine. I also checked the surrounding points to see if I had any tin sticking out from underneath the board and shorting something out and that also seems to be fine. 

I saw that the version 
xLoader update 06/26/2022
It already fixed the driver issues with Ape Escape is it possible that the latest FW does not include this fix?

ramapcsx2, 2024-02-26:
The PAL versions of the game all have libcrypt, and if that triggers, the pad won't work.
Can you tell me which version of the game exactly you have? (PAL-E, or PAL-G, PAL-I etc). Maybe it's missing in the database.

PSXmanCZ, 2024-02-26:
Thank you, after so many years I learned something new again. I'm used to using the Original CD. 

I have the Ape Escape version SLES-01564 if I understood it correctly by studying on the internet so it should be PAL-E.

I noticed that you can also download "sbi files" as patches for the emulator on the internet. Can this file be read by xstation?

I am thinking that if I have a pal console the best will use also PAL versions of games and as I see probably not exactly. Because the more games I run I find that more games have a problem with the analog mod but for most it doesn't bother to play and just turn it off.

ramapcsx2, 2024-02-26:
Oki, I checked and the ID is properly included. The game works here for me.
The way you describe the error clearly indicates that libcrypt triggered though, which is a bit puzzling to me.
If you're running any recent firmware (it has been included for years now), it should work normally.
Maybe the game has been altered?

PSXmanCZ, 2024-02-26:
You are right, I downloaded another image from the same SLES disk. And it works the original version I had from archive.org. I apologize and thank you for your time. Now I have to find out why my BlueRetro adapter is not working correctly. I may have to look for a newer revision of the SCPH-5552 console.

closed 2024-02-26 by PSXmanCZ