x-station/xstation-issues #345: play homebrew games

fabiopallini, 2024-02-28:
Hello, I am developing a homebrew game for PS1. Currently, the game has been tested on various ps1 models without xstation installed: SCPH-100x, SCPH-900x, SCPH-700x, and SCPH-500x, and it's working properly on these. However, when my friend tried to play it on a PS1 with XStation, he encountered only a black screen, he tried to launch the game with FAST LOAD and FULL LOAD. I'm wondering if there's another method to run homebrew games on this system or if there might be another underlying issue, maybe I'm missing something.

The PS1 with XStation is a PAL system, and it's functioning normally with 'official' games loaded from the SD card.

ramapcsx2, 2024-02-28:
Hey, if you could pitch me a build (can be stripped down if needed), I can check it out.
It is expected that every game works, with only a handful of exceptions.
That the game runs from CD-R but not xStation is very unusual.

fabiopallini, 2024-02-28:
You can download the demo (game.iso) directly from this link:
https://github.com/fabiopallini/ps1-graphics-demo/releases
There are various versions available. Try version 0.6.3, or version 0.6.1, which is the latest version I have tested on real hardware (700x + action replay).
Thanks.

ramapcsx2, 2024-02-28:
https://i.imgur.com/P9TFVGu.jpg
Looks good?

It might still be a problem with the current release firmware. I gave it a test on the development version here, but it booted right away.

ramapcsx2, 2024-02-28:
[image]
Tip: Generally don't use GetStat like this. It clogs up the Mech with useless extra work, which can make it miss important events. You know your data is ready when you receive the INT1.

ramapcsx2, 2024-02-28:
I also tried the earlier 0.61 version. This one has CD mastering errors that xStation automatically fixes. This function was present in release firmwares for a while, but maybe it missed the bad sectors here, not sure.
This CD mastering error is gone in 0.63 btw.
[image]

ramapcsx2, 2024-02-28:
Excellent song choice in beta 63 with Bust :D

fabiopallini, 2024-02-28:
Yeah, it's working. The songs are just for testing :D 
Maybe there's a problem with the firmware. I think my friend has version 2.0.2. 
The 0.6.3 version has a different build system. I built that version with buildcd, I think that's the reason.
Anyway, thanks for testing this, much appreciated

ramapcsx2, 2024-02-28:
Sure, and good luck with your game! :)

Some info on the mastering problem:
When you have errors in the CD creator, and then burn this image to CD-R, the burn software will silently fix the problems.
It is automatically patching up problems, so the CD-R then works fine.
XStation has such repair functionality as well, but it doesn't have enough information to reliably detect every possible fault.
The detection is quite good, but it's possible that this exact flaw was unfixed in the current release firmware.
It is however definitely working in the next version :)

fabiopallini, 2024-02-28:
Thank you very much for the clarification