x-station/xstation-issues #346: View game cover art in the game list menu

AzagraMac, 2024-03-02:
The possibility to insert game covers and have them displayed in the list menu. 

It would be possible to make a file structure, where inside a folder the covers are put, putting as name the ID of the game, or that directly from internet I download it from 

PAL: https://psxdatacenter.com/pal_list.html

NTSC-U: https://psxdatacenter.com/ntsc-u_list.html

NSTC-J:
https://psxdatacenter.com/ntsc-j_list.html

or 

https://raw.githubusercontent.com/xlenore/psx-covers/main/covers/default/${serial}.jpg

ramapcsx2, 2024-03-02:
Hey,
Covers are a big feature that are requested occasionally. This is a project where I'll need community help.
To get started, first the loader's code base, which is Unirom, needs to be open-sourced.
The Unirom creator has essentially wrapped up the code cleanups required for the release, but it hasn't happened yet (real life getting in the way).
When it is done though, we can team up and start thinking about how to do covers :)

AzagraMac, 2024-03-03:
The information is appreciated. 

A menu, with a little information about the game, region, and cover, would give it a very powerful look. 

I offer to collaborate as much as possible to contribute something to the project if possible. 

regards