x-station/xstation-issues #347: Late pu8 issue when the lid is closed

gnarl84, 2024-03-06:
I've installed many of these units but have not seen this issue yet. The console will only boot when the lid is open and when you close the lid or manually press the switch down it shorts out and keeps trying to reboot itself. I've triple checked all my solder joints and placed electrical tape over them to be sure they're not contacting the rf shielding on the bottom.

gnarl84, 2024-03-06:
I should also mention I installed the qsb on a pu18 board and have 2 ribbon cables but am getting the same results. Could this be an issue with the main xstation board?

ramapcsx2, 2024-03-06:
Hey, it's not likely to be a fault on the xStation, but it could be many other things.
First of all, are there any other mods installed?
Second, when you say you installed < the qsb > also in a PU-18, I don't fully understand what you mean there.
Do you mean to say that the same xStation board gives the same problem when using in a PU-18?

gnarl84, 2024-03-06:
Sorry I should have clarified. I have a late pu8 as well as a pu18 and have installed a qsb into both consoles. After assembling each system I am seeing what looks like a short with the lid closed.

ramapcsx2, 2024-03-07:
Alright, got it. That now does look like a fault :p 
I'm always mentioning the robustness of the xStation itself first, because the in field fault rate (that i know of) is like .. 3 or so :)

If you have a multimeter on hand, you could probe whether the line that carries the reset signal is shorted to ground or Vcc on the xStation.
Another place this could go bad is the flat flex connector, and there is an "R2" and "R3" resistor array on the xStation that could bridge pins.
It's still pretty weird but a multimeter beep test will probably show a short.

Depending on that, you may be able to repair it, but either way, you can always contact the seller and ask for a replacement.
Please try to find the fault first though... but if you want to return it, you can say i sent you :p

gnarl84, 2024-03-07:
Thank you for the suggestions I'll check those out tonight!

gnarl84, 2024-03-12:
Update - I decided to try a third console  (another scph 5501) and am still having issues. I was able to boot to the xstation, update firmware and even load a game for around 10 minutes to test. I powered off the console and came back a couple hours later and its now only booting to the cd/mem card screen. I've never had this many issues with multiple consoles so I'll be trying to exchange the unit

closed 2024-03-12 by gnarl84