x-station/xstation-issues #348: Programs for using cheats on firmware [2.0.2]

Sseerii, 2024-03-07:
Hi all. Xstation is still one of the best mods for PlayStation, thanks for that. And thank you for solving the problems that have arisen. Now I'll ask a question...

I want to use cheat codes in games, but it doesn't work. I don't have a separate device, so I only test CD formats. My tests and their results:
1) GameShark CDX - cheats are not saved in the game. None of the versions of the programs are in my photo: [image]
2) GameShark Light - cheats are not saved in the game.
3) Xploder 9000 Version 2 - works in the game Doom, but does not work in Resident Evil 3, the screen begins to flicker and freeze:
[image]
4) Xploder 9000 Version 4 - the game Doom freezes on the splash screen, Gran Turismo 1 cheats are activated, but the music disappears and freezes when selecting a garage.
5) The collection of programs in CHEAT_ENGINE_COMPILATION does not work, cheats are either not saved (Codebreaker) or games freeze (Xploder 9000 all versions). The instructions say that emulators are not supported, but Xstation is not an emulator, so I don’t understand why that is...

Thus, cheats can only be used if the save file is saved on the memory card (Probably Game Guru), which is completely inconvenient. I am guided by logic: Select disk 1 with Cheats, select disk 2 with the game, launch the disk with cheats (Quickly or completely does not affect performance), activate cheats, open and close the lid, check in the game.

I saw a video where it works, but as I understand it, this is one of the first firmwares, the main menu is too different: https://www.youtube.com/watch?v=yYBrfvRLu48&t=379s&ab_channel=Chris%27sRetroGaming

For example, here a person stated that cheats using this method do not work in firmware 1.4.5: https://www.youtube.com/watch?v=pKlfFr6v4SY&ab_channel=SIGARETTE

I don't think that changing the firmware to an older one is the right decision in this regard. Unlike PSIO, here we can use an external device via a parallel port, but I have not gotten to this point yet. Please tell me what we can do now with this firmware. Thank you!

ramapcsx2, 2024-03-08:
Hey,
there is no reason this shouldn't work, no matter the xStation firmware.
The problem is probably with the cheat CD. Can you try some others?

Sseerii, 2024-03-08:
Okay, I'll come back with this question in a couple of days, when I've tested all the programs in the cheat engine compilation.

I hope that cheat users will post in this thread and perhaps we will be able to compile a small rating of programs.
[image]

ramapcsx2, 2024-03-09:
As I said, there is really no reason that this shouldn't work. I don't know what you guys are doing there :p

Sseerii, 2024-03-23:
Hello. I'll come back with information.
In my experience, unfortunately, codes do not work without additional devices. Only GameShark with the ability to create a separate save...

You definitely need a separate device to work with Gameshark. The SD version tries to apply cheats, but as I wrote above, the games start to work incorrectly and freeze.

As soon as I connected a separate device, everything worked. Code Cracker 9000, which was in Russia, also works on the same principle.

My result is this. What a blessing that the parallel port is free, unlike PSIO. Another advantage :)

closed 2024-03-23 by Sseerii