x-station/xstation-issues #349: Compatible micro sd cards

pirsir76, 2024-03-11:
Hello again.

I asked the question before if 1.5TB micro SD cards will be compatible.
I'm specifically talking about this one: https://www.bhphotovideo.com/c/product/1792744-REG/sandisk_sdsquac_1t50_gn6ma_1_5tb_ultra_uhs_i_microsd.html

I do recall ramapcsx2 saying "The default storage format is microSD, with sizes up to 1TB tested, and this is likely to be the maximum."
Is there any specific reason why 1TB is likely the maximum?

I actually just bought the 1.5TB Sandisk but I have asked them to hold it just in case.
I'm willing to test it, but I would like to hear from ramapscx2 first about how sure he is about it being able to work? If there's even a small chance of it working I'm willing to test out the 1.5TB Sandisk.

ramapcsx2, 2024-03-12:
Hey ho,
someone else on Discord also just ordered a 1.5TB card, test results expected soon :)

The absolute limit (at which the address space runs out) is 2TB, but with the last bit being the deciding factor, it's uncertain whether the software stack allows for the entire range. We'll see.

pirsir76, 2024-03-13:
"someone else on Discord also just ordered a 1.5TB card"
Can you share the discord link please?
All the ones I've tried end up being invalid.

pirsir76, 2024-04-16:
Just to let others know; yes the new SanDisk 1.5TB Ultra microSDXC works 100% with the x-station.
It has 1.39TB of available space on it.
Downside is the write speed is very low; approximately 20MB/s. So, expect it to be slow when you are copying games onto the micro SD. (There is no known way around the slow write speeds.)

I will close this topic now.

closed 2024-04-16 by pirsir76