x-station/xstation-issues #350: Keeps corrupting gamescache.txt and config.txt

Recoil816, 2024-03-16:
So, I have had my xstation installed for quite awhile, and I've updated to the most recent firmware as of today.

I can have a handful of games (anywhere from a few to maybe a dozen or so) on the SD card at one time.  But once I get to a certain arbitrary amount, and go to "refresh game list", it will "work" for a minute, and then jump to the screen saying that no games can be found.  And from that point on, it will no longer work.  I can delete every game off the card except one, and the issue will still occur upon booting it.  I've already tried putting them into subfolders and everything.  I've formatted the card multiple times, checked to make sure everything is named correctly, etc. etc.

One thing I keep noticing is in the 00xstation folder, config.txt and gamescache.txt become corrupted and cannot be deleted manually after this error occurs.  I check the details of those two files and the "Last Date Modified" has changed on both of them to 12/31/1979 11:00pm".

It's the weirdest thing.  Like I said, it works perfectly with a few games on there.  But once I get to about 20 games or so, give or take, this always happens, and the only way to fix it is to format the card yet again and start over.  I've tried putting entirely different games on it too, just to rule out a corrupted one.  Doesn't make a difference.  Anyone have any idea?

ramapcsx2, 2024-03-16:
I do :)
It sounds like the card is damaged, or maybe you use an SD card extender?
When refreshing the game list, the xStation looks at ever file and folder in order, while writing down (into games.txt and gamescache.txt) all playable things.
If the process fails with an SD card write error, there will be corrupted files now and the file system in limbo.
Even that should typically not prevent games from running (ie you enable folder based browsing).
Since it still breaks then though, my best guess is a bad SD card.

Recoil816, 2024-03-17:
Thanks for the reply.

Well, I've tried a different card too.  It does the exact same thing.  See, if it's only about 10 games or so, it has literally zero issues.  But if you dare add too many more than that, it just goes off the rails.  My brother has his as well that he is going to bring over in a few days.  Going to try my SD card, loaded with a couple dozen games on his.  And also his SD card on mine.  Trying to see if *somehow* it could be a hardware issue in the console or in the xstation install itself.  Hopefully not.

ramapcsx2, 2024-03-17:
I need to know whether you use an SD card extender, those with a long wire on them and a small adapter PCB.
What you're describing is loss of signal on the SD bus, which could be a bad card, or the effect of the extender, or indeed a hardware problem on the xStation.
For it to be a hardware problem, the SD slot might have damaged pins, or otherwise there's a poor connection between the SD card and the ESP microcontroller.

On another note, does it work normally if you use folder based browsing (which doesn't need to refresh the game list, as it doesn't use it)?

Recoil816, 2024-03-17:
Nope.  No extender.  I actually haven't tried to use folder-based browsing.  I guess I'll give that a shot and see what it does.

ramapcsx2, 2024-03-18:
The curious thing is that the problem occurs after filling the card over some mark.
I think the card(s) are suspect. Would be best to test with a few more.