x-station/xstation-issues #352: Horned Owl (Japan) Stuck on Copyright Text

RaouleMenard, 2024-03-19:
Hi,

I have a Japanese Fat PS1 with XStation 2.0.2, playing on a CRT via composite cable.

The lightgun game "Horned Owl" - Japanese version gets stuck at the start, on the first screen with "Sony Entertainment..." white text on black background.

The same dump works fine on another PSone with a PSIO.

The game only launch with fast boot. When full boot the screen flicker and go crazy.

Any chance for a fix? Thanks!

ramapcsx2, 2024-03-19:
Hey, 
I just tested the game on 2 consoles here, and it's fine on both, in full and fast boot.

Can you tell me which model PSX you have?
And could you install firmware 1.6.1 to check whether the game works there?

RaouleMenard, 2024-03-19:
> Hey, I just tested the game on 2 consoles here, and it's fine on both, in full and fast boot.
> 
> Can you tell me which model PSX you have? And could you install firmware 1.6.1 to check whether the game works there?

Thanks for your reply.

I'm not sure about the console version, it's a Japanese Fat PSX (SCPH-5500). Got it on ebay with the xStation already installed.

However, I've installed firmware 1.6.1 (while keeping the latest loader) and now the game works fine!

Thanks for your help!

closed 2024-03-19 by RaouleMenard

ramapcsx2, 2024-03-19:
Okay, yours is a PU-18 for reference :)
The game works on my PU-18 here, so maybe there is a problem in the release 2.0.2 firmware that's fixed in my local dev branch.
I'll keep the game on a test list.