x-station/xstation-issues #354: continually corrupts sd cards

eveunlimited, 2024-03-28:
installed the xstation into a late pu-8 1001 model. after a bit of troubleshooting (just weird sd card formatting issues), the xstation worked fine. it was booting into the menu fine, it recognized a game, but i didnt run it as i was working on it at work and wanted to go home.
i got it home, took the card out, added some more games, plugged it back in, and i'm getting the blinking green light signifying no card. putting it back into the pc, the entire card is corrupted and unrecoverable. multiple trusted formatting tools can't save it, all it can recognize is the size of the card. it's happened to two cards, both a sandisk and a microcenter brand. the install is fine and the board did work, just now any sd card inserted seems to die on me.

eveunlimited, 2024-03-28:
oops, it was my own fault. should've troubleshooted more. i was using a usb hub that never gave me issues with miscrosd before, but i plugged into io on my pc itself and its back to working properly. thanks for the good product and stressful solder job!

closed 2024-03-28 by eveunlimited

ramapcsx2, 2024-03-28:
Yep, I'm using hubs also, and reliability between cables, adapters and sheer luck is spotty :p 
Enjoy the thing! :)