x-station/xstation-issues #355: Most games consistently freezing after selection

Honda76, 2024-04-11:
Hi,

First time installing an Xstation in an SCPH-5502 PU-18 board and had to first remove a modchip that was already there but was a quick easy task. I should also say that the unit had the usual laser issues but would play CDs fine before the install.
However, games discs would not boot at all.

The xstation install was finished yesterday and everything seemed fine. Latest firmware 2.0.2 was installed and all good. 
Xstation Loader menu boots successfully everytime, but most games chosen freeze moments after selecting them in the Loader!

Have a variety of games on the sd, some US some EU but all work fine on RetroPie.
Out of about 25 games, only 3 work fully: Bomberman (EU), R-Type Delta (EU) and Spyro (EU)

Tekken 3 starts and you get to choose a character but then freezes when you select.
Destruction Derby you hear audio but then freezes moments later.
A few others including Final Doom look like they are about to run but freeze on or just after the initial Turqoise text.
The rest of the games (including all Crash Bandicoot and Wipeout) all freeze just after the Loader screen disappears. The screen stays black.

All the problems are consistent. Nothing is intermittent. Choosing full or fast load makes no difference.

I have checked the QSB and all solder points look good and continuity of all pins 1-9 and 11-20 is ok. This includes pin12 all the way back to the point where the resistor is removed.
I checked the lifted pins and flowed the pads underneath IC701 to check for shorts.

SD card is not Sandisk but is a brand new Kingston Canvas Plus A1 128GB and was formatted exFAT by the official sd card formatter.
I have ordered a new Sandisk Extreme 256GB micro sd but just wanted to ask if this sounds like an sd card issue or more to do with the game files themselves?
Assuming corruption is not an issue as they all play fine on RetroPie.

ramapcsx2, 2024-04-11:
Hey,
behaviour like that suggests severe SD card issues, but with a slim chance that the soldering is not 100% or that things like an SD extender or a bad slot are causing it.
My guess is a poor SD card.

Honda76, 2024-04-11:
Thx for the reply. 

No SD extender is in use. Just the xstation.
All other xstation functionality seems fine including reset.

SanDisk arriving tomorrow, so will update thread after I've tried again with new card.

ramapcsx2, 2024-04-12:
Forgot to say yesterday, but the big update is already in beta, over at Discord.
It has many SD card related improvements and could get even yours working. If you want to give it a go:
https://discord.gg/8ykTpzUfEk

Honda76, 2024-04-12:
Yeah I think I might try the beta.

New SanDisk Extreme had made no difference. 
Only games that run are the same ones.

But I think I might have realised that this particular pu-18 board might have a pre-existing issue.
I never actually owned a ps1 originally, but wanted to give it a go getting one and getting it working again, so I have no experience knowing if what it is doing is correct or not.
I tried fixing the laser unit but quickly decided it wasn't worth the effort so decided to go with xstation.
But every time I power it on (and also before xstation install) the startup screen with either 'SCEE' or 'SCEA' would often be (and often still is) either '[KME' or '[KMM' ? Is this normal for a ps1 with a modchip inside it? 
Also, I'm getting no in game music. I think R-Type Delta has music but none plays. FX are fine. 

I'm on the discord now. Where do I find the beta firmware?

Honda76, 2024-04-12:
Ahh i found your pinned message and have the 2.1.0 firmware now. I will give this a go and report back.

Honda76, 2024-04-12:
OK, so new firmware 2.1.0 seems to be making no difference at all.
Same games play,  but the rest do not.

I beginning to think this is a dodgy pu-18 board.
Below is an image I took of the modchip install, after i removed the tape around it but before I removed it completely.
It does not look correct when compared to other installs images online and am I am wondering if the original modchip install has caused some damage to the components.
[image]

ramapcsx2, 2024-04-12:
"either '[KME' or '[KMM' ? Is this normal for a ps1 with a modchip inside it?"
Nope, found the problem :p 

This PU-18s Decoder IC might be bad. This can only be fixed by swapping the "1815Q" IC, but I'd rather go luck for a different board..