x-station/xstation-issues #356: is it possible to edit the games list in the xstation menu?

kaitlin459, 2024-04-15:
so i got my xstation sd card set up with the games i want. im using cheat engine and was wondering if there a way to edit the menu so that the cheat engine iso is at the top of the list. having to scroll down the list every time i wanan use cheat engine sux id rather it be at the top so i can just click on it and load it from there without having to search the list for it thanks for your time

ramapcsx2, 2024-04-15:
Hey,
at this time, you cannot disable the automatic sorting.
As a workaround, you could rename your cheat CD so that it starts with a "0", making it go to the top of the list.

I want to implement a favourites list, as well as a history of the last played games, but it will take a while.

kaitlin459, 2024-04-15:
@ramapcsx2 any chance you can link me to a guide on how to edit my cheat cd? i got the iso from the internet archive but idk how to edit a ps1 cd/iso

kaitlin459, 2024-04-15:
tried to edit the iso with cdmage and it still shows up under my c folder so idk what im doin wrong i just want the cheat engine compilation bin file to show up at the top of my list so its easier to use it

ramapcsx2, 2024-04-15:
If you you have cheat_cd.bin and cheat_cd.cue (and nothing else), then rename as such:
- 0cheat_cd.bin
- cheat_cd.cue
Then open cheat_cd.cue and rename the entry in it from "cheat_cd.bin" to "0cheat_cd.bin".

kaitlin459, 2024-04-15:
i got it sorted also if you wanna add this into your readme so people dont have to ask the cheat engine compilation iso does indeed work with the memcardpro tested it with the action replay both with and without the memcardpro inserted and with it inserted it had no issues saving to the sd card