x-station/xstation-issues #357: xstation not booting game unless i reset the console

kaitlin459, 2024-04-16:
so bought a premodded ps1 scph 1001 the one with the red white and yellow rca jacks the xstation boots to the menu but when i select a game and press either fast load or full load it shows a black screen that says xstation the name of the chosen game then loaded system.CNF but then in order to load the game i have to press the reset button to load the game any ideas on whats causing this?

otherwise she runs the games fine just sux having to press reset to load each game

[image]

ramapcsx2, 2024-04-16:
There might be a problem with the reset line, but I'm not sure...
The normal way to tell is that when you select a game and pick "full boot", it will not automatically reset the console.
But when you select "fast boot", there is no reset step intended. The game should load and smoothly start.

kaitlin459, 2024-04-16:
well when i select full boot what do i do after the black screen where it says X game loaded system.CNF? once the Xstation get to the image seen in the pic below what do i do to play the game? if it matters the ingame reset device is not installed

kaitlin459, 2024-04-16:
well when i select full boot what do i do after the black screen where it says X game loaded system.CNF? once the Xstation get to the image seen in the pic below what do i do to play the game? if it matters the ingame reset device is not installed

[image]

ramapcsx2, 2024-04-16:
Fast boot / pressing X on the game should show the screen you show and then immediately switch to the game (no reset to the whole startup sequence).
If that doesn't happen for you, then something is wrong.

kaitlin459, 2024-04-16:
but if i do full boot and i get the screen below what do i do? if i choose fast boot the game loads no problem but if i do full boot the game get to the image below and sits there
[image]

ramapcsx2, 2024-04-17:
Okay, thank you. So fast boot works and that is good :)

Since full boot is not working, there is most likely a problem with the reset signal towards xStation.
It can't trigger the signal to reset the console, and likely it also can't read it.
Check the soldering around this area:
https://github.com/x-station/xstation-issues?tab=readme-ov-file#xstation-cant-reset-the-console-full-boot-not-working-after-updates-no-automatic-reset

kaitlin459, 2024-04-17:
since i dont have the soldering skills im just gonna leave it alone as pressing the reset button to load the game when using full boot isnt to much of a hassel

ramapcsx2, 2024-04-17:
Alright. If you had the mod done, then you can surely ask the modder to take a look and fix it.
Otherwise, maybe ask around a bit locally, or find trusted modders online... if it ever starts to get annoying that is :)

kaitlin459, 2024-04-17:
yea so far she still works fine i just have to press the reset button after pressing full boot and i can tolerate that thanks for the help

my2k2zx2, 2024-05-02:
> yea so far she still works fine i just have to press the reset button after pressing full boot and i can tolerate that thanks for the help

Sounds like a missing/poor connection on the qsb to the reset line.  If you are in the US, I would fix this at no charge for you.

kaitlin459, 2024-05-02:
@my2k2zx2 i am in arizona do you have discord or an email so we can discuss this further one guy quoted me 80$ just to look at it to see if he could fix it that wasnt even including shipping costs?

my2k2zx2, 2024-05-02:
my name here is the same on discord.  :)

ramapcsx2, 2024-05-02:
Oh gee, I understand if modders make their work a business and have to cover for all sorts of issues, but a thing like this is better handled by someone nice enough to do it as more of a gift / nice thing to do :p

kaitlin459, 2024-05-02:
@my2k2zx2 heres my discord nixiepixie6922

kaitlin459, 2024-05-02:
@ramapcsx2 he wasnt a modder who had a business he was just some guy who did a few for his friends then saw my post in a forum about my issues and sent me a pm offering to do it for 80

my2k2zx2, 2024-06-24:
@ramapcsx2   You can close this issue.  Point 17 of the qsb was not connected.  Once repaired, the full boot option is now working.

ramapcsx2, 2024-06-25:
Thanks again for fixing people's consoles, super nice move! :)

closed 2024-06-25 by ramapcsx2