x-station/xstation-issues #358: tf card is not recognized when boot. Prompts to put in the correct disc

gcb2008, 2024-04-25:


gcb2008, 2024-04-25:
5500 Japanese version

ramapcsx2, 2024-04-25:
Please create a short video that shows the problem. I need to see the xStation LEDs when the console powers up.

gcb2008, 2024-04-26:
> 请创建一个简短的视频显示问题。当控制台启动时,我需要看到X站LED。

I am unable to provide a video..It's not my machine, my friend's He has replaced the new xstation. According to his description, the SONY logo appears when the machine is turned on. Then the blue background prompts to put in the correct PlayStation disc. Take out the tf card to enter the memory card management interface.

ramapcsx2, 2024-04-26:
Usually, the problem is with the QSB soldering.
If you check old issues here, many people report this as their problem, and sometimes it gets fixed.
I think it's best if you translate those issues and try to follow the advice that worked.
(Another option is finding an installer, who can check the work for you.)

https://github.com/x-station/xstation-issues/issues/323
https://github.com/x-station/xstation-issues/issues/283
https://github.com/x-station/xstation-issues/issues/339

gcb2008, 2024-04-26:
> Usually, the problem is with the QSB soldering. If you check old issues here, many people report this as their problem, and sometimes it gets fixed. I think it's best if you translate those issues [...]

Thank you for your reply