x-station/xstation-issues #359: FF Chronicles FFIV stutter in Towns/Castle/Dungeon but not in the world map?

Ledoeye, 2024-05-05:
Hi! So yesterday i started a new play through of FFIV on my Xstation PS1 and noticed right of the bat in the beginning castle there is some scrolling stutter motion going on not sure if it's related to the port itself of this game or some compatibility issue with the xstation, i only experience this in town/castle/dungeon tho, world map is fine.  

I've played numerous titles on my Xstation Ps1 and never once experienced an issue so I feel like it has to be the game itself but just would like to know for sure. What confuses me is that this does not happen on an emulator as i tried it on duckstation with no issue. 
For anyone here that has played this version of FFIV did you notice the same stutter? Thanks in advance.

ramapcsx2, 2024-05-06:
Hey ho,
I didn't play this version of the game, but we can very likely rule out the ODE as causing problems here.
The game runs through a peculiar kind of emulator, where framerate issues are not unusual.
You may want to confirm that you're running the NTSC version of the game, and ideally on an NTSC console, matching the 60Hz ideal rate.

Ledoeye, 2024-05-06:
Had no idea it was running through an emulator o.o But yeah i'm running the NTSC version. I will be closing this issue since it's clear it has nothing to do with the xstation, thanks!

ramapcsx2, 2024-05-06:
All those games are quite sad, with the excessive load times and such. Sadly, I can't do anything for them in xStation. They just need to load a ton of data.. (especially Chrono Trigger)