x-station/xstation-issues #361: Dance Dance Revolution

Kingbrian16, 2024-05-17:
Hello!

Noticed an issue with the Dance Dance Revolution Series in Japanese and English. They don't boot fully and are stuck on the loading screen. Is there any updates on the new firmware that would fix the issue by any chance? A workaround is reverting to a previous firmware but by reverting will that mess anything up?

Thanks!

ramapcsx2, 2024-05-17:
Hey,
you can revert to 1.6.1 without worry, if you want to.
Alternatively, the open beta seems to be working nicely. You can find it here on Discord:
https://discord.gg/e2r74R5hx7

Kingbrian16, 2024-05-17:
Thanks! Sorry for all the questions, but I have a couple more. Does the beta have any issues that will mess anything up? I was reading the chat and did not pick up on any major issues, would it be safe to update to this Beta version? Or do you recommend 1.6.1?

ramapcsx2, 2024-05-17:
From what people report, the beta is fine. 
I'm recommend being cautious and reading the chat, but this beta has been out for a month now, and all feedback so far was positive. It's gonna be okay :p