x-station/xstation-issues #363: Xstation no pasa del logotipo de sony con la sd puesta.

ivansusanna, 2024-05-29:
Hola.

Eh echo una instlacion en una ps1 pu-8 japonesa y cuando intento iniciar con una sd puesta en el xstation no pasa del logotipo de sony pantalla blanca, pero si arranco sin la sd sale el menu del cd con una pista que si le doy empieza a reproducirse pero no se escuha nada.
Eh revisado la instlaion varias veces y comprovado continuidad en los puntos y todo esta bien pero na hay manera de que salga el menu del xstation cada vez que le pongo la sd. Las sd que eh probado son sandisk de diferentes capacidades.

ivansusanna, 2024-05-29:
Lo unico raro que noto es que en 2 condensadores al tomar medicion me da continuidad y no se si eso es normar, adjunto foto marcando en rojo los condensadores que le pasa.
[image]

ramapcsx2, 2024-05-30:
Those 2 resistors are low ohms (22Ohm), so reading continuity is fine on them.

I'm sorry, I can't really think of a case where the white Sony screen gets stuck forever, that occurs when trying to read a CD.
I think the xStation mostly works in this system, but some kind of error prevents reading data.
It might be that the decoder chip in the system is damaged, which sadly happens sometimes.

I don't really know what to suggest now, other than trying to find a PU-18 board and installing the xStation there..

ivansusanna, 2024-05-31:
Hola.
Gracias por tu tiempo y en contestar, creo que al final es lo que voy hacer intentar conseguir una placa PU-18 por que el xStation funciona perfectamente lo eh podido probar en otra consola con con una instalacion ya echa y funcionando, asi que el problema como dices debe de estar en esa placa que algo no esta bien y hace que no pase de la pantalla blanca de Sony.

ramapcsx2, 2024-06-01:
Okay, I'm sorry I can't really do more here. But PU-8 boards are known to have elevated numbers of bad Decoder, Decoder RAM, SPU, SPU RAM chips. It is likely the cause.
Good luck with the PU-18 :)

ivansusanna, 2024-06-02:
Gracias por la ayuda, ya eh instalado en una placa PU-18 japonesa y funcionando todo perfectamente a la primera, tendre en cuenta en futuras instalaciones el tema de las placas PU-8 intentando evitarlas, gracias por todo otra vez y un saludo.

closed 2024-06-02 by ivansusanna