x-station/xstation-issues #365: Audio stutter eventually followed by no music audio and not booting when the console gets warm

Gothrocker, 2024-06-21:
Hi, I've installed the mod on a stock NTSC-J PU18 board.

Works perfectly fine when booting the console cold, but after like half an hour to an hour or so when it gets warm I get audio stutters at first, then after ~5 minutes only the game music disappears while other sounds are still playing, and if I reset the console at that point I get a distorted boot sound, followed by the xStation menu not booting. It boots into the memory card menu/cd player screen instead. When the console cools down it works fine again for some time.

I thought it would be a QSB solder point issue, since I have issues only when the console gets hot, although I've reflowed each solder point and tested all of them for continuity and bridging, and they seem fine.

I've tried different firmware versions and SD cards (formatted according to the guide) and I'm not using an SD card extender either. During the install I did lift pin 55 instead of 56 at first on accident, but I fixed it after. Looking at the IC701 pins, I don't see any bridging pins or any that aren't making a connection. Not sure how to for continuity/bridging on the IC701 pins though.

Is there any specific pin on the QSB that could be causing this specific issue that I should be focusing on? Or maybe it could be caused by pin 55 on the IC701? Not sure if something unrelated to the xStation install could be causing it either. Unfortunately when I got the console I only tested it for like 10 minutes with some random CD, so I can't confirm if the issue existed beforehand.

Any advice is appreciated.

ramapcsx2, 2024-06-22:
Hey,
focus on the bad SPU boot audio here. This can't be caused by any xStation signal, so I think maybe you lifted SPU pins by accident and reconnected them? (And then check those again, if so :p)

Gothrocker, 2024-06-22:
Hi, thanks for the reply.

Googling for SPU gives me CXD2925Q for PU-18 boards, and isn't that a separate chip from IC701 (CXD2545Q)? I've 100% only touched pins on the IC701, so you're saying the problem was possibly pre-existing, and that it could still cause the xStation not too boot (only when hot)?

ramapcsx2, 2024-06-22:
The chip is correct, yes. It seems to occur quite often that people accidentally lift pins on the SPU, instead of the DSP, and issues such as yours might develop if the repair of that was not properly done. It's a fault i hear about, so I mentioned it here.

So you haven't touched the SPU, but still have that behaviour of SPU audio being distorted...
It might be that your machine has some chip damage, and it'd be around the SPU then, maybe SPU RAM.
It can develop by itself / out of the blue, but personally, I haven't seen that happen yet.

So what to do next:
Check SPU and SPU RAM area for possible problems. Try touching / pressing down on them in operation, see whether audio becomes weird from that, indicating a cold solder joint..

Gothrocker, 2024-06-22:
Thanks for taking the time to help and sorry for the wall of text.

Just to clarify, in case anyone with the same issue would be reading this in the future, the audio is only distorted during the intro screen when I restart, after the issues start appearing and it doesn't boot until it cools down. Before that it works fine until the console gets hot, then it begins by stuttering music, followed by all sounds playing fine but the only music disappearing (no distortion here, if that even means anything). I can continue playing the game fine without music until I restart and it doesn't boot. Gets fixed when the console cools down and it breaks when it gets warm.

I did check the SPU chip and it looks fine. Everything looks clean and none of the pins move when I touch them with a needle, so they should be making contact. I've had the console open and playing for over an hour now and pressing down on any of the chips doesn't do anything, but with the console open I can't even replicate the issue. I can only guess now it's not getting hot enough.

I've seen in youtube videos some models having a copper looking heat shield over the chips on the right side of the board, which neither of my two 5500 consoles have. If that's causing it to get too hot to operate then I don't even know. It's not like it gets too hot to the touch either way.

Since it's a Japanese console and I'm in Europe I did put a RePSX PSU, although it doesn't get hot and I'm using a good quality external brick. The xStation board gets a little hot too, but I'm sure it's not to the point where it's a problem, and I'm also using it's standoffs. It's not like the console is sitting on a carpet or something either.

I thought it could also be pin 55 on the DSP, since I did lift it on accident and soldered it back down again, even though I don't know what that connects to either way. Figured there could be too much solder causing a bridge when it gets hot. Used wick and a magnifying glass on it and the surrounding pins. Tested for bridging. It all seems fine.

I do have another PU18 board I could reinstall the mod to but...

Also reading this thread below regarding alleged heat issues and you talking about the lid switch possibly being affected by heat? It really is heat I guess I could try that new PicoPSU mod with the noctua fan...

https://github.com/x-station/xstation-issues/issues/214

Gothrocker, 2024-06-22:
So I just installed the mod to a different PU-18 board and the same thing happens. It just overheats. Doesn't seem to be a console issue to me at this point. 

I touched the xStation board while the console was on this time and it was super hot allover. Not sure if that's normal but cooling it down with a blow dryer fixes it until it gets hot again.

ramapcsx2, 2024-07-02:
Note for people finding this issue later: 
I've received this particular xStation after a debug session on Discord couldn't find the fault.
Will now check it thoroughly and hopefully find what could be wrong with it..

Gothrocker, 2024-08-10:
Long story short, the xStation was getting too hot for some reason. Rama helped me diagnose my install, which seemed fine, then I sent the unit to him. He was very helpful and in the end Dragonbox were kind enough to send me a new unit, which works great now. I also installed Noctua fan using my RePSX PSU just to have. I hear new firmware will also improve things once it's ready.

Thanks a lot to Rama and Dragonbox again.

Gothrocker, 2024-08-10:
Resolved with Rama's help.

closed 2024-08-10 by Gothrocker

ramapcsx2, 2024-08-11:
Ah, cheers for updating :)

The unit is strange indeed. There is no fault at all with it, except that it heats up more than usual, to the point where in a warm room, I can see it glitching out.
I suspect that there are some ESP32 modules out there that may have been cut from the lowest quality silicon, yet still passed QA, or maybe the flash chip is marginal.

Seeing this for myself helped, insofar that I'll double down on the next firmware's power saving.
At some -40% power, even this module will never overheat.