x-station/xstation-issues #366: Black Screen After Quick Boot

twistedsymphony, 2024-07-01:
I recently Installed an X-Station (Rev 1.2a) in my SCPH-1001 (Late PU-8 model). Before the install the console worked but had some issues reading Discs (good use case for an ODE). The install went smoothly. I have a lot of soldering experience with modchips and other ODEs and a good digital soldering station so I wasn't at all concerned about the QSB.

I got the Xstation booted, I put 3 games on my SD card including Einhander and Wipeout. I did a full boot of Einhander, played for a bit. then. I updated the firmware to the latest from the repo (2.0.2). The firmware updated on screen said it finished and I watched the console reboot after completion.  And the Menu showed the proper new firmware version.

After this I attempted I launch Wipeout and selected quick boot rather than full boot figuring I'd see what the differences were.  The console went to a black screen and stayed there. I waited a good 2 or 3 minutes and then shut down and turned back on... and got a complete black screen.  no logos or sounds on boot. The same thing if I remove the SD card, and the same thing if I unplug the Xstation from the FFC.

I check over the QSB with the image here (https://github.com/x-station/xstation-issues) and confirmed continuity with <0.5ohm resistance between the FFC pin and the test point on the mobo for all FFC pins, confirmed no shorts. also confirmed no shorts and good continuity on adjacent pins from the lifted pins on the to board. For sanity I even reflowed all of the pins on the QSB, and when that had no joy I removed the QCB to look for shorts underneath (none were found), cleaned off all solder and re-installed. Here are some pictures of the board as it sits now: https://photos.app.goo.gl/u92hFjAJbDBp7Usv5 

I'm at a loss as to what to try now. if anyone has any further troubleshooting steps I'd love to hear them.

ramapcsx2, 2024-07-01:
Hey, that's highly curious error behaviour.
The update clearly installed and works, as you were able to restart into the loader and select a game.
Using fastboot is completely trivial in regards to the hardware, so the fault wasn't caused by that.
I believe you have a defective ESP32, which worked for a little while, but died soon after the update.. Maybe the flash on it was damaged, and it went unnoticed at production.

I think it's best to contact the seller and ask for options. In cases like this, the go to solution is a free replacement.

ramapcsx2, 2024-07-01:
Please tell me one more thing though:
When you power the console on, does the green LED do anything at all? Please try this with and without an SD card inserted.
The blue light is working, right?

twistedsymphony, 2024-07-01:
With SD card inserted the blue light comes on with console power and I get one quick flash from the Green LED. it does this whether or not the Lid switch is open or closed.

With SD card removed the blue light comes on with the console power and the Green LED flashes slowly (about .5 sec on, then .5 sec off). It does this whether or not the Lid switch is open or closed.

ramapcsx2, 2024-07-01:
Ok, hmm, all reverse, the ESP32 is working fine, the firmware is running :p 

So I guess something else changed then, maybe a signal doesn't make it to their destination anymore.
If you'd like, you could check the R2 and R3 resistor arrays (near the flat flex connector), as well as the single resistor in the middle.
It might also be a bad cable or a bad connector, reseating the flat flex might show that.

twistedsymphony, 2024-07-01:
I've re-seated that FFC probably a dozen times by now, but I haven't checked the cable itself. I'll try that and see if I have any replacements kicking around. I'll check out the resistor arrays too, thanks.

ramapcsx2, 2024-07-01:
Sure. And if you want, you can hop on Discord for more realtime help effort. We'll see what we can do :)

https://discord.gg/EbjJpTt4H3

twistedsymphony, 2024-07-01:
I've confirmed the resistor arrays are good as is the cable and connectors. I'll check out the discord for further troubleshooting

closed 2024-07-03 by twistedsymphony