x-station/xstation-issues #367: Alternative to M.O.D.E. Installation Query

Sisee84, 2024-07-01:
My enquiry isn't an issue, but a question regarding the installation and compatibility of the XStation.

I have already modified my PlayStation with the Terraonion MODE kit.
My question is; do I need to remove the MODE board from my console... which is potentially hazardous given how it is installed... to be able to use the XStation?
Or is the board the MODE uses compatible with the XStation?

I don't really have access to another console, and removing the soldered in MODE board is a difficult task given my tools, so I'd like to avoid it if possible.
Although I like the low-cost storage the MODE can use in SATA drives, but the cost of the XStation itself is lower.
Just to clarify; I use my MODE for my PlayStation, Saturn and Dreamcast, but would like to have permanent installed devices to make life more convenient... which is exactly why these kinds of devices exist in the first place.

If the XStation is not compatible with the MODE's PSX Installation Kit board then I need to consider weather the cost of another console plus a XStation kit is better than purchasing a second MODE unit, as my estimates put those prices pretty close to eachother.
I really do want to avoid removing the MODE board I soldered into my PlayStation, as it was fiddly and difficult to do.

Thanks.

ramapcsx2, 2024-07-01:
Hey,
the MODE QSB is most certainly not compatible, eventhough it's "inspired" by xStation's. Still a bit sour on that but eh :p 

You're right in the removal being tricky and risky part. It would be best if that console stays a MODE unit.
A conversion is doable, but tricky work that you need good tools and experience for.

So ultimately, if you want to try the xStation, it is just best to go look for a new board. 
PU-18 recommended, though PU-8 (LATE) is fine, and the earlier ones are only rudimentary supported (lack of buyable QSBs.)

I understand that price is a concern, but that all the more speaks for keeping the MODE PSX as is, I think.
They didn't ask me about pinouts, so nothing I can do now :p

Sisee84, 2024-07-01:
Thanks for the reply.
I thought there may be some proprietary stuff in the MODE board, even though it is suspiciously similar to the XStation one.
It was a long-shot, but I was kind of hoping there was just a simple adapter or something.
I'll have to take some time to think about how I want to go forward now.
Like I said on my opening pist; a second console plus the XStation kit is looking to cost me a similar price to a second MODE on its own with the current state of the market.
I'm not committed yet, as as much as the inconvenience is to dismantle my consoles to move the unit from one to another is... I don't actually so so all that often.
Partly because of the effort it takes, and also I am trying to limit the wear-and-tear on the various connections and cables.

Thanks again for the reply.
It makes me happy that people like you are so open with the community.

ramapcsx2, 2024-07-01:
No worries, but please update me if you pick a choice eventually. Curious now :)

Sisee84, 2024-07-01:
I'll try.
Although I wasn't planning anything in the immediate future... I asked the question because its been a curiosity since I first installed the MODE board in my console a few years ago now.

Another curiosity;
If I were to install the XStation; is it easily reversible?
As in; can the board be left in-place but use the optical drive like a stock console.
Or is it a 'permanent' modification to the console?
I ask because the MODE is the latter, in that the optical drive is completely unusable without removing the board and reconnecting some pins that needed to be lifted on the consoles chips.
Basically; a massive chore with many risks involved.

In an ideal world I'd love for there to be a PSone compatible ODE, as I adore those cute little consoles.
But, to my knowledge there isn't one... yet.

Like I said;
If and when I decide the hassle of transplanting the MODE from one console to another becomes irritating enough for me to buy anything to make the installations more permanent... I'll try to let you know.
I am kind of curious about the alternatives to what I have now though, so may end up getting a second PlayStation and a XStation. But, I'm not making any promises at this time.

ramapcsx2, 2024-07-02:
It's a bit about sensible compromises when designing a PSX low level ODE, which xStation and MODE are.
(They intercept DSP commands, such as jumps or spindle kick / brake, etc, and not the "high level" game commands, such as "seek to sector XYZ, which is what PSIO does..)

A low level ODE requires disabling, bypassing and rerouting many internal signals, which is due to the PSX being designed as it is.
In return, this ODE type can achieve very high compatibility, by keeping most of the timing/behaviour crucial hardware untouched.

When designing this, there was a choice on whether to keep it "simple", which is what xStation/MODE do, and loose access to the original drive, or go elaborate (and expensive).

The elaborate option was simply too much for me to <also> consider back then, but it would have been an extra board with switch buffers, through which the user can pick between real drive and ODE.
This is doable in general, but a bit tricky now, as an afterthought.

In the end, I decided to go straight for the basics, and we now have a still affordable, very compatible ODE, since I could focus all energy in making the core product solid.
The fact that there's over 100 million of these in the wild also helped with the decision.

I hope this rant gives you at least one dev's perspective on this stuff, and thanks for listening! :)

Sisee84, 2024-07-02:
I didn't read that as a rant.
What I see is someone who is passionate about something.
And while some of the terms you used did sail over my understanding of these things; its clear to me that you know your stuff.
I consider myself a competent electrician and mechanical engineer with a decent grasp on the digital realm beyond the average persons; your understanding of the functionality of the components and how to manipulate them is impressive to me.

Also, the fact that you haven't responded to me with blunt and basic statements, and even took the time to 'rant' about the subject...
Well, that shows me that you not only care about the subject, but the people who you expect to bene fit from your work.
What may have started out as a personal project for yourself has become your contribution to the world, and I get the impression that you haven't done that for profit... although you do still need to pay the bills... but because you want others to benefit from your work.
It is a somewhat noble act.

For the record I spent a hour on eBay last night; looking at PlayStation consoles.
I'm still not committing yet, as I am sure you would understand that life tends to get in the way of the best laid plans.
But, I am thinking of rescuing a console that has abandoned the ability to read CDs... as I'd be removing the laser anyway; I feel that not only will I be reviving a console that others would throw away or use for spare parts, but saving myself a considerable amount of money too.
XStation-ising a PlayStation is something I may end up doing at some point... even if just to satiate my curiosity.
Although, it isn't a pressing need in my life at this moment and I'm not looking for a project right now.
If and when I do decide to; I'll come back and ask you for your recommendations and input again... if that's alright with you.

It really has been a pleasure talking with you, and I hope you continue to get the pleasure from your passions in the future.

And a little recommendation from someone familiar with other things;
Don't let the politics and less honourable practices of others get you down.
I know there's a lot of competition and not everyone plays fairly by the unspoken rules others do.
So long as you can take pride in your work and do so knowing you aren't abusing the ignorance of those you offer said work to... you can sleep at night knowing you are a good person.

Thank you very much for the discussion.
It's far more personal than when I tried to contact Terraonion about an issue with the MODE... which a lot of other users have and still are experiencing because it has never been fixed.
If you are curious I'm referring to the crackling sound issue, which only seems to be from audio streamed from the disc as through it was originally Red Book format... if I have the standard correct.
I noticed it when playing Tomb Raider and reported it straight away, which was acknowledged as a 'known' issue... and never resolved.
If I am honest; although my MODE works fine... other than the issue I just mentioned... the lack of continued support on the product has left me feeling a little upset.
The promises of continued updates and support seem to have halted around the time I got my unit, and I have given up checking for any progress to be honest.
Now I'm having a rant of my own.
It's not that I am not happy with my MODE; it does exactly what I need it to.
I just think that the creator seems to not care as much about their existing products and focussing far too much on future ones.
I have found a similar mentality from other well known people who offer similar products for retro consoles such as flash-carts.
It seems once they finish one product it's out the door and onto the next.
I understand that's how business works, but it isn't very customer friendly... especially when a company is a small number of people or even just one who isn't able to dedicate the attention each of their projects require.

Thank you for enduring my rant.

ramapcsx2, 2024-07-03:
Man.. it took me a while to read, comprehend, and respond here.
Thank you for your nice words. :)

I've talked with the developer of the MODE software core, neodev.
He's a great and humble guy, and I got the clear impression that he spent a lot of time on figuring out what the console needs.
I knew about the problems MODE had at the time, and gave him some hints on how to address them.
The fixes are somewhat difficult to implement and test, and the music crackling problem cannot be addressed (in software) at all.
There's a chance that fix attempts with not enough testing time will make it worse overall.

Now, the other thing here is that he's part of a larger team, who are a company, and need to put earning money high on their priorities, or else they'd go under.
This puts the man in a position where he probably cannot spend enough time on perfecting things, and so this is why I think the PSX portion hasn't seen any updates.

Though, given all that, it's been a few years now. Maybe an update might be good by now, at least for the correctable things.
Maybe someone should poke them on it :p 

Again, I want to thank you for your words. 
It's great to hear someone getting this impression from my stuff. It's a great motivator here :)