x-station/xstation-issues #368: Freeze

Rexarrrr, 2024-07-08:
When listening to music in the road rash music track menu, the console freezes. Freezing occurs randomly, the picture is static and does not respond to buttons. It doesn't freeze immediately, after 30 minutes - an hour of listening. The problem has been noticed twice already.

Rexarrrr, 2024-07-08:
I've been using xstation for a year now and have launched more than a hundred games without any similar problems.

ramapcsx2, 2024-07-08:
Hey,
the game is buggy. It will freeze from CD also :p

ramapcsx2, 2024-07-08:
Testing "Road Rash" now, in the Jukebox player, I've got 2 things:
You say you run this for 30 minutes or more, but for me, it always auto exits to demo mode. Any tricks?
And second, wow, it crashes quite often on selecting a track. Silence and dead game. In the logs, there is no CD activity even, suggesting another really buggy game? :p

ramapcsx2, 2024-07-08:
Yea, there is no CD Pause, or Seek, or anything at all on the CD side. It's going to be another EA production :p

ramapcsx2, 2024-07-08:
Selecting this title has a pretty high chance of hanging the game. The previous song stops, but no command is ever issued to the CD section. It just becomes dead .. 
[image]

Rexarrrr, 2024-07-08:
Thnx)

ramapcsx2, 2024-07-08:
Oh this isn't done. I'm still investigating :p
Will update if I find anything, but so far it looks like it's going to be one of those games that also hang from real CD.
All the EA "Road" games have a pretty bad code base. They're known for sluggish menus and weird crashes.