x-station/xstation-issues #370: No Boot screen / Black screen / No blue led

oddzandgodz, 2024-07-20:
Hello, 

I have a problem with my PS1 SCPH-5552. I made a clean install 2 years ago of an xstation, everything was working perfectly until few days ago when I turned off normally my console and it never worked again. The console power supply works, I have the green light of the console but all I get is a black screen, no boot screen with the PS1 white page and logo. I checked several video cables, but it's not the problem. I opened my console, and realized the xstation led doesn't light up at all, no blue or green light. I have made several voltage measurements, but I don't know what voltage I should get on the Xstation. All I get on the J3 connector is 3.4V on several pins but no 7V or 5V. So I guess I have a faulty component somewhere but don't know where to search.

ramapcsx2, 2024-07-23:
Hey,
I was out a few days, sorry for the late response.

You would want to measure the 3 legs of the little voltage regulator. One is +8V, one is 3.3V, the last is Ground.
When measuring, you should connect one of your probes to the ESP32 module metal can for an easy Ground.

Since you don't have a blue light though, I suppose the 8V fuse on the PSX mainboard is blown.

oddzandgodz, 2024-07-23:
Something really weird happened. I couldn't find anything wrong on the mainboard, so I decided with despair to plug it and test it again and it worked ! Then I switched off the console and now it's not working again. So it may be a bad contact or a faulty component ?

ramapcsx2, 2024-07-23:
Okay, best take off the flat flex cable to the xStation and carefully clean both contact ends.
Ideally you have some isoprop alcohol for this. You want to wipe off some oxidation here.
Once done, reassemble and try again, see if it's any better.