x-station/xstation-issues #371: Xstation blocked on PlayStation white logo

brakvader, 2024-07-31:
Hi,
My xstation won't boot, it flash one time green led then stay blocked on PlayStation white screen. If i open the lid, goes on cd menu, and if i close, no cd find. I've checked my continuity with bip and the schema on mmonkey UK. It's a japanese 5500 pu18. I've tryed to change sd card. And the xstation well create the 00xstation folder and config the first Time, so it's writting on the sd card. Need help please.

kalm-traveler, 2024-08-03:
Did you install it yourself? 
Did it work previously and suddenly quit working?
Can you post pictures of the top and bottom of the mainboard? 

Need all the details to try to help!

ramapcsx2, 2024-08-03:
Oh, I missed this, sorry!
Yea, need some more info please :)

brakvader, 2024-08-03:
Hi, 
Oh yes sure, i'm in vacation, i'll be back 10 August. So will do pictures.
I install it myself. Never worked. 
And just look at continuity that i Saw on the mmonkey UK. Led blue ok, led green flash once and nothing After. Try 2 sd card scandisk.
When i Come back i update with pictures.
Thanks !

ramapcsx2, 2024-08-03:
Okay, so far what you describe is the generic QSB not quite soldered right case. Most people with issues have this.
There are several things important here, for example the fact that measuring continuity easily connects loose solder joints from the probe pressure. It will read fine, when it is in fact a cold solder joint.
Another problem could be a damaged Decoder IC on the PSX. If the Decoder can't decode data, or doesn't reply properly at boot, the system could hang on the white Sony screen.
Yet another issue could be excess flux use, where the remaining flux conducts just enough to bridge 2 signals together, causing a quasi short..
So yeah, ideally, hop on Discord for some live support. I'll give you a fresh link when you're back :)

brakvader, 2024-08-11:
Yup,
I'm back. So i checked my continuity again, m'y pin 5 and 13 are connected , is it normal ? Or is it the point...
[image]
[image]
[image]

ramapcsx2, 2024-08-11:
It doesn't look good, gotta say. Please check with your meter that this point here is not connected to ground:
[image]

https://discord.gg/AmuWaQeC

brakvader, 2024-08-11:
Not grounded, the picture is not super quality, it's better on real.

brakvader, 2024-08-11:
So, on the header pin 5 and pin 13 was biping. Was not the qsb, but 2 pads of the dsp was in contact. It's donne, working perfectly, thanks again.

closed 2024-08-11 by brakvader