x-station/xstation-issues #372: Memory Cards copy functionality - can't copy save game file if existing file with same first few characters exists

kalm-traveler, 2024-08-03:
Just finished setting up a 5501 for myself and I've got about a dozen backups of my old memory cards from back in the day (thank you Dex Drive) which I've been trying to copy onto pages of a MemCard Pro 2 via the XStation's "Memory Cards" function on a given game selection of the game menu.

This is very fast for organizing them into the correct memory card folders since the Xstation triggers the MCP2 to switch to the corresponding game with Game ID, but I've run into an odd behavior.

Across these old memory card images, there are quite a few instances where there are save game files which either have the exact same 'name' or simply have the same first few characters such as "FF7/SAVE01/<different time>" but the Xstation memory cards screen won't allow me to copy the save file with an error that a file with that name already exists.

I _can_ copy them with the default memory card manager. This is just a bit of hassle since that won't auto switch between MCP2 cards via Game ID, and to get to that menu I need to disconnect the cheat cart and reboot with the lid open.

Just curious why the Xstation memory cards manager behaves like this, especially since many games do not even mark on their save files which slot a file was created in.

[image]

ramapcsx2, 2024-08-03:
Can you load this FF7 "SAVE01" correctly in the game?
I wonder what the game even does, since it now sees 2x the same ID in slot 1.

kalm-traveler, 2024-08-03:
ah it ignores the 2nd file. For FF7 there's a PC utility you can use to edit all the properties of save slots so I suppose I could combine everything into one and then use that to change the save slot number in the title.

So far I've noticed this behavior with saves for FF7 and 8, Lunar Silver Star Story Complete, and Resident Evil 2 Dual Shock edition. 

I guess the only simple solution is to copy these slot-specific saves to separate vmcs?

Just on a glance, it seems maybe some games use the identifier and/or title but not all. RE2 does as well as the Final Fantasy games as seen here: 
[image]

but Castlevania Symphony of the Night does not - in game I can see both of the save files that begin with CASTLEVANIA-1 and have DRAX00 identifier:
[image]

ramapcsx2, 2024-08-03:
I see. 
This means that games can pick what they use as identifier, and my guess is that most will go with the shorthand ID one, while some also evaluate the "title" one.
I'll have to chat with @JonathanDotCel, see what he thinks is best to do here.
For now though, it's probably best to keep those same "ID" saves on separate VMCs.

kalm-traveler, 2024-08-03:
Sounds good and thanks! I was just surprised by the behavior since the internal memory card manager doesn't have that restriction, and it doesn't impact 'all' games.

I suppose for games that don't have a specific fan-made save modification utility application, if one just _had_ to consolidate overlapping slot save files, one could load the offending name save from one card, then in-game-save on a different card's free slot eh?

ramapcsx2, 2024-08-03:
Right, and the same thing is doable for external tools. Maybe there's an option?