x-station/xstation-issues #373: Full load does not load games

HasdelBora, 2024-08-10:
Fast load is OK but full load freeze the console.

ramapcsx2, 2024-08-11:
Hey,
that sounds like the xStation doesn't have the system Reset signal properly. 
To quickly verify this, you could do a firmware upgrade, and see if the final step resets the console properly.
If that step looks glitched, or the console keeps displaying the update information forever, the reset signal is not working right.

If it's a PU-18, please check your QSB install around point 17 here:
https://github.com/x-station/xstation-issues?tab=readme-ov-file#pu-18

Check point 17 here for PU-8:
https://github.com/x-station/xstation-issues?tab=readme-ov-file#late-pu-8--21-and-later

HasdelBora, 2024-08-11:
I read that clone of pico PSU can be the problem... because of missing reset circuit... Do you agree that?

ramapcsx2, 2024-08-11:
The reset circuit on the PSU is required for the PSX to function correctly, that's right.
This being completely missing is <bad>, but check your PSU again carefully. It might be there.
It would be a tiny, often 3 leg part, somewhere center, or nearby the reset button.

As a test to confirm or rule out this as the problem, please try it with the original PSU installed.

HasdelBora, 2024-08-11:
Point 17 tested and looks fine... Update doesn't reset my console, I have to reset with the reset button... (Even my xreset doesn't work when the console freezes) Think my Pico PSU clone is the guilt one 🤣 Don't have OG one to test

ramapcsx2, 2024-08-11:
So yea, problem is the reset signal, for one or the other reason, not sure atm. You need to find what that problem is, and xStation should work properly again.

HasdelBora, 2024-08-11:
I ordered an OG PSU on eBay... For testing... I will let u know here :) Maybe I will order a Will's Mod one later... But fucking expensive from France :(