x-station/xstation-issues #375: PS1 Resets sometimes mid game, power supply issue? Xstation?

Ledoeye, 2024-08-22:
Hello! I bought a PS1 like 3 weeks ago, tested it for a bit and the only thing that seemed wrong with it was that the laser was bad so I modded it with the X-station and recapped the whole thing including PSU.
Games run beautifully with no issues, the problem is that the Ps1 sometimes resets on it's own. It's random, yesterday for example i played for like 5 hours and not a single reset.

I've done 2 things so far, reflowed with fresh solder the 2 main power solder points and i've also reflowed the reset switch but that's about it, issue still persists and like i said above it's random, sometimes it happens other times it doesn't.
Could this just be a bad power supply? Or an X-station issue? Thanks in advance.

ramapcsx2, 2024-08-22:
Hey,
it's going to be a bit tricky to debug, but if you're methodical, should be fixable.

Resets occur if the PSX reset line gets activated (bad PSU reset button, bad PSU power switch, causing the reset circuit to kick in, etc).
Try to power down the console and remove the mains plug. Then push the reset and power buttons repeatedly for many times (maybe 50x+), so to clear them of debris.

The xStation has the ability to assert the reset line also, but it will only do that on a special command, from within the loader.
It's unlikely the xStation or its installation have to do with it, though you may want to check the reset line soldering.
For PU-18, this is about signal "17" here: https://github.com/x-station/xstation-issues?tab=readme-ov-file#pu-18

Ledoeye, 2024-08-22:
Hey @ramapcsx2 just followed your instructions, i went as far as completely de soldering both the power / reset switch and then cleaning em really well with contact cleaner. Also reflowed a couple more points like the 5 pin connector coming from the motherboard / PSU. 

The console has been powered on for almost 5 hours and a half hours without resetting itself, i will wait like 2 more days and update the post 👍Thanks a lot!

closed 2024-08-25 by Ledoeye

ramapcsx2, 2024-08-25:
Seems to be good? It's a somewhat common thing with these consoles :p