x-station/xstation-issues #377: Part of QSB missing, any fix possible?

Krilllon, 2024-09-05:
Hello, 

So installed the mod using the official guide. Assumed the qsb was correct as the one shown in the guide was v1a and i had v1b. But as it turns out the bottom left bit is missing a piece...

Anyway to fix this without getting another qsb?

Thanks in advance 
[image]

ramapcsx2, 2024-09-06:
Ah, I'm really sorry about that. It's a weak point of the design, where the corner can break off in shipping / handling..

So what's missing now is the very important main grounding, as well as that same ground connected to a nearby signal to pull it permanently down. See here:
[image]

In addition, where the break occurred, there's still a big grounding lane on the QSB that is now open and exposed. It might make accidental contact with signal 20, so you should check (multimeter) that it doesn't.

Once you've cleared that, you can repair this with patch wires.
The big intended ground spot is freely accessible, so you solder a mid-sized wire to this, and the other end to the flat flex cable connector at either its pin 1, or just the mounting metal at the side of it. I've marked these:
[image]

Finally, you just connect the other signal to that ground as well:
[image]

Krilllon, 2024-09-07:
Hi, thanks for the detailed explanation. So done all of those things, but now just get a blank screen when I turn on the PSX...

Also noticed that no LEDs light up on the actual xstation board.

Is there anyway to test that the actual board isn't dead?

ramapcsx2, 2024-09-07:
So you probably don't actually have ground. No current is flowing, hence no LED, hence no picture.
You may assume the xStation is fine, they are robust.

Krilllon, 2024-09-13:
Hi, sorry for the delay in responding. Took the "easy" way out and ordered another QSB.

PS1 boots fine, goes to the default PS1 menu. Tried the CD test and works. But still can't get it to boot into actual XStation.

If the CD test works, does it mean the flex cable is connected properly at least?

[image]

While testing for continuity, there is continiuty between all of the points in the picture. Is that expected or is that the issue?

ramapcsx2, 2024-09-13:
Hmm, sorry for the bad experience :/
When the CD test works, insofar that a silent CD plays, with the seconds counter ticking up, that means almost all things are working correctly.
However, there is still an issue where the format of the data sent is not right, typically because there are shorts or "quasi-shorts" on the I2S signals:
[image]