x-station/xstation-issues #378: PU-18 Full boot issues

Darkone83, 2024-09-06:
I have a PU-18 SCPH-5501 

X-Station has been installed for some time and works fine for fast boot, just not full boot

Reset point is tested and has continuity, added jumper wire makes no difference

I do have a Bitfunx PS1PSU installed but the behavior is the same with the OEM PSU

I have reflowed everything on the QSB and validated continuity on all test points

Last firmware that Full boot seemed to work on was 1.4.4 currently 2.0.2 is installed

Any ideas?

ramapcsx2, 2024-09-06:
Hey,
you seem to have tested every common fault point for these already, but yea, the xStation appears to not be able to assert the reset signal.
This could be caused by a cold solder joint on the QSB, which you may have tested in a way that puts pressure on the solder point, closing the contact on the reading.
If an older firmware works, but not the latest ones, then it looks more so like an intermittent signal path.
Other firmwares tend to have slightly different settings on how the signal is pulled down, which could make it kind of work then.
It should however be fine on all of them, of course :)

Darkone83, 2024-09-06:
Only light pressure used to ensure it was actually testing the joint as expected. It's an oddity but otherwise, it performs amazingly well. It was only something that always made me go hmm. Now if I full boot it will constantly reset but if I manually reset the system it will boot the loaded disc. So not the end of the world to me :)

ramapcsx2, 2024-09-06:
So the system is generally simple:
The Reset line in the PSX is broadly connected to all the chips, and when this is low, the chips will be in reset.
The power supply has a small IC that ensures the reset assert pulse is consistent for some 200ms, no glitches etc on it.
XStation can also assert this, and will also do it for about 200ms.
This reset pulse has to be glitch free, or there won't be a reset at all, or possibly some half-done midway state.

This signal is an input on every chip, and and output on xStation or the PSU whenever they want to reset, otherwise also "input".
It is also pulled up by a pullup resistor on the mainboard. Nothing else on it.

So if this is glitched, the most likely cause is that the signal path is interrupted. Maybe the flat flex cable has worn contacts, or some other issue like that, but it should be a "plain" fix :p

Darkone83, 2024-09-06:
Gotcha I noticed there is no reset circuit on the Bitfunx psu only a transistor (IE no IC003) so that's likely my issue then. I'll order a US PSU and test with that to verify. But ty for the input

ramapcsx2, 2024-09-06:
Ah gee, yea, I can't really tell what effect these corner cutting PSU designs will have on it. It could even be dangerous, in case the PSU output is <<not>> tristated as it should be.

Darkone83, 2024-09-06:
Understandable, power output is stable and I have been using it that way for years. I think the idea was is not many people would leverage that signal or subsystem so it's easier to cut power as a reset method temporarily. While it might not be the best method it's crude and effective. But I went for one of those so as not to stress out the JP PSU that was in there as I'm on a US power system. But ill doublecheck when I can get the chance and report back.

ramapcsx2, 2024-09-06:
Oh it cuts all power? That is super bad for the console. The chips need a synchronized reset pulse to reliably start in sync.
It will likely work anyway, but it's not designed that way..

Darkone83, 2024-09-06:
Lol agreed I might need to see if it can "rebuild" the circuit as a mod we shall see

Darkone83, 2024-09-06:
I'm speculating it's because M51957B isn't present to talk to to trigger the reset and that's why it's not happy

ramapcsx2, 2024-09-06:
Think about the system as whole. 
The PSU might be missing the reset control circuit, but that would not change how xStation asserts reset for say, 200ms... as long as the PSU doesn't do something bad on its control of this Reset line.
What would be bad for example was if the PSU always drove Reset to High...
It would mean when xStation wants to reset (drive Reset low), the 2 would fight it out, creating a shortcut for 200ms.