x-station/xstation-issues #379: Xstation PU-18 Install / Blue LED Only / Black Screen on Boot.

HarryFlowers, 2024-09-12:
I was hoping to get some guidance on a specific issue I've been getting with my Xstation installs. I purchased an Xstation which came with two QSB's for both 100x and 500x models. My first attempt was on a Late PU-8 (SCPH-1001). Continuity tests checked out, even to the main Xstation PCB board with the flex cable plugged in, but after booting the PSX... the screen flickered then stayed black (no white Sony logo or Dark PS Logo splash). There was only a blue LED that would stay on the main Xstation board while the PSX power was on, no green light at all.

I figured I might've just  created a short somewhere under the QSB, and had another 100x model lying around, so desoldered the QSB and did it again there... same exact issue. In trying to reflow my solder points to troubleshoot... I accidentally burned off some copper on one of the soldering points on the QSB. So I decided to purchase a working 500x model and try the other QSB.

The SCPH-5501 was in completely working condition (could boot games just fine from the optical drive), so I went ahead and installed the 500x QSB. Again, continuity tests worked out perfectly fine from the mother board all the way to the main Xstation ribbon socket with the flex ribbon connected. The same issue persisted, only an initial flicker on the CRT followed by a black screen after pressing powering on.

Out of desperation and sheer frustration to get this installed, I purchased another SCPH-5501 (working condition), removed the QSB from the previous one, cleaned it thoroughly, and installed it once more on my 4th Playstation this time. Continuity points worked perfectly (again), but the exact same issue persisted.

At this stage I'm beginning to think the problem is not on my end, seeing as how my soldering work is good, and continuity has been confirmed from the motherboard all the way to the Xstation board (with Flex cable plugged in). I find it hard to believe that I would run into the exact same issue four times in a row by way of an install error...

Could you help provide some information that I could use to see if there is potentially a faulty piece from the Xstation I received?

There is no green LED that appears whatsoever on the Xstation board, with or without an SD card inserted. Would this point to an issue with the main Xstation board rather than either of the QSB's?

ramapcsx2, 2024-09-13:
Sounds .. fishy, yea.
The persistent issue seems to be that on startup, there is some video, but only briefly, and distorted.
It suggests that the APLL clock signal that drives the PSX CPU is not working right.
This may be due to various defects, for example, try to measure the single resistor on the xStation flat flex area that says "R4".
It should read about 56 Ohms.
It's very safe to assume that whatever fault there is, it's with the xStation parts (xStation PCB, flat flex cable, QSB), possibly with the APLL wire (check it's not getting grounded by the shell / metal shield).

It would be great to see the APLL signal on an oscilloscope, but I don't think you have one ready, right? :p 

If you cannot seem to get it working, contact the seller and ask for a replacement. They're lenient with this and should send you a new one.

ramapcsx2, 2024-09-13:
One more: There never was any SD card extenders involved, right? Those long cable ones from China? That crush the card slot and cause shorts?

HarryFlowers, 2024-09-13:
Firstly just wanted to say thanks for the timely response, very much appreciated. However, I believe my situation may have gone from dire to worse...

To answer some of your questions:

1. No never had an SD card extender attached, I was waiting until I had it working until I looked into one of those.
2. Unfortunately do not have an oscilloscope.
3. I have Kapton tape covering all of the soldering points on the underside of the PU-8 motherboard, so I would doubt that the wire or any other solder would be making direct contact with the shielding.
4. Unfortunately I've run out of time tonight to dissasemble the PSX to take photos of the soldering, but when I have a chance tomorrow I can do so.

Now onto the latest developments...

I got my multimeter out and tested R4, resistance came out to 56.2 Ohms:

[image]

While I was reading your response, I found a similar issue from a previous user, and you had suggested to measure the voltage of the voltage regulator of the Xstation PCB. Prior to taking photos of it, the voltages read out to 7.8V on the one, and 3.'something' (can't remember exactly) on the other, so pretty much where it's supposed to be.

However, when I attempted to do it again and have a photo of it, it appears I may have shorted something out? My black probe was always touching the ESP32 metal top, but as soon as my red probe touched one of the voltage regulator pins... the Blue LED turned off. I've tired unplugging the power cable form the console, re-seating both the flex cable and the Xstation PCB... and now no matter what I do I can't get the blue LED to light up.

[image]

My questions now are:

1. If my voltage / resistance readings were initially coming up correctly, would that have been an indicator of something being wrong with the Xstation PCB, or the installation?
2. Now that I can seem to get power to the PCB, does that mean the whole thing is cooked?

ramapcsx2, 2024-09-13:
1: It would have meant those things are okay, the issue would be elsewhere. Process of elimination :p
2: I guess the only 2 ways to zap the thing with a multimeter would be if the meter was not in Voltage mode, or if you slipped with the probes and bridged 2 pins on the regular. My guess is #2

Such a short would probably have blown one of the PSX fuses, and to repair this, you should swap it out against a spare.
Fuses are around the PSX PSU connector, marked PS601 etc.
[image]
(The row of 3 x 20 likes to blow on these issues.)

Fixing this will restore power, and usually no further harm was done.
It still remains that you get no solid video, eventhough the xStation appears to be working, or at least initial checks looked okay on it.
We need to know how come that the video is garbled on startup... Any small solder balls etc on the xStation, or damaged flat flex cable might do that, but it's tough without a scope to check anything.

ramapcsx2, 2024-09-13:
One note: Please don't measure resistance with power enabled. This is done with all power off.

HarryFlowers, 2024-09-13:
So I checked the fuses near the power connector, and fuse PS601 seemed to have failed. I wasn't getting any reading for resistance, and no continuity. The rest of the fuses were good. 

I went ahead and took a 20 from my spare x5501 and replaced the blown one. I then put everything back together and when I powered it on I got the same result (no Blue LED). At first I thought my soldering could've been bad, so I did it again with another 20 from the spare x5501... same thing, and then a third and final time with the 3rd 20 from the spare x5501. I realized that since I'd been testing the fuse after replacing it, and was getting a resistance reading of 0.2-0.4 ohms, as well as a passing continuity check... the fuse was getting blown each time I would attempt to power on the Playstation.

I didn't think to record or take a photo of my multimeter checks after replacing the fuse each time, but here are some photos of the install:

[image]
[image]
[image]

I decided to put a pin in troubleshooting this part, to know that for some reason the Xstation PCB was just shorting the PS601 fuse every time I powered on the Playstation, I decided to pull out the last x1001 model that still have the QSB connected to the mother board. This time, however, I did take a video of testing the fuses before powering it on, and then after. 

After transferring the Xstation PCB to the x1001 model, re-seating it and hooking everything back up, I went to power it on and indeed... same result: Playstation powers on, but no blue LED to the Xstation. I opened it up and tested the fuses one last time, and same behavior: PS601 was not working. 

Below is the video of me testing the x1001 fuses before and after powering it on with the Xstation installed:

https://github.com/user-attachments/assets/a6fa5d6f-7fd0-4124-a655-37b80d8c2081

So it seems my problems have gotten a bit worse here... how bad is it exactly? Have you seen this behavior before where the main PCB seems to short fuse PS106 every time? I can confirm that everything in both models are properly installed and grounded... I'm just scratching my head at this point.

ramapcsx2, 2024-09-13:
Oh gee, so one thing you learn with experience is that if a replaced fuse blows again right away, you now know for certain that there is a short, and you first need to fix that, before attempting any more :p

So your xStation has a short now, pretty sure, but I couldn't say from here where it might be.
It happened when you probed the 3 pin voltage regulator, so maybe the part failed internally, but it could be something else as well.

What I really recommend at this point is that you ask the seller for a replacement xStation, but then you also send one otherwise working console and this new xStation to a mod shop, and have it installed.
It can happen that with repeat frustrated installation failures, one locks themselves into never getting it right, and if you want to avoid that, this is the way.

ramapcsx2, 2024-09-13:
And to add to that, I also think that this xStation had a fault from the start, where it would never fully work, causing those glitched video outputs. So a replacement is fair.

HarryFlowers, 2024-09-14:
Ah you're probably right...

I'll reach out to the seller and describe what my experience was and hope they're in a good mood! The box it came it wasn't the typical 'black' Xstation box I've seen in videos and images, just a generic carboard box with the pieces individually placed in plastic bags. I'm not sure if that's normal, or maybe they had been RMA'd previously.

I'd been making a lot of great progress learning circuitry and soldering (several other console mods, guitar pedals, some entry-level robotics, etc...). This was the first project that gave me any real issues, so I'm somewhat determined to get it right! However, as you said I'm also going to take a step back from this, cool my head, and see if I can get it done from a more experienced modder (thinking of reaching out to Voultar).

Thanks for all of the help, very informative and educational none the less. If I ever figure out what was causing issues with this one I'll make sure to update the thread in case the information is useful.

ramapcsx2, 2024-09-14:
:)
The box it comes in is up to my partner's discretion. Personally, I favour something super plain, no ink on it, no stickers, nothing that's going to become waste more often than not..., but I'm no marketing expert either :p

A defective unit that someone sent back either goes to me (so I might check what went wrong with it), or it would be tested and verified working normally, then it might be used again (but I don't know how they handle it).
It can happen that someone is unhappy with the thing, and just want their money back. I'd imagine this is when a working unit would be sent back. Anything defective would be replaced, so there's no point, right?

When you want to try again yourself, the best suggestion I can give is that you lightly press down the QSB when soldering it.
You want it flush to the PSX board, so solder can easily bridge the distance. This should entirely prevent accidental bridges etc.