x-station/xstation-issues #380: Xstation PU-18 Install / I'm only getting the Memory Card/ Music Player screen after update firmware

rdasilva84, 2024-09-15:
I received the Xstation and sent it to a trusted service center for installation.
After the firmware update, the boot no longer starts on the Xstation, it only goes to the Memory Card/ Music Player screen.
The technician recorded a video demonstrating the problem and carrying out the test on two consoles.

[https://youtu.be/Hk_L83-YVmw](url)

Could you help me?

ramapcsx2, 2024-09-16:
Hey,
so this worked once, enough to run a firmware update, then never again. Is that right?

The videos only tell me that the xStation is confused, probably not receiving or processing control commands.
Things like that after a firmware update could mean various things, for example a previously stronger signal output is now weaker, messing up operation. The cause then would be a bad connection to the QSB (both of them), so maybe an issue with the resistor arrays on the xStation... but I can only speculate really.

Please drop by the Discord channel later, I'll work you through the firmware downgrade process. You probably want to know whether that fixes it :p
https://discord.gg/EbjJpTt4H3

rdasilva84, 2024-09-16:
I asked the technician to send me the console, since his workshop is in another city. As soon as it arrives I will contact you so we can carry out the tests. Thank you very much.

ramapcsx2, 2024-09-17:
Sure, please ping me on Discord then.
But expect a little debugging session. This kind of problem so far was never resolved with a firmware downgrade. There is an underlying issue that we need to fix :)

ramapcsx2, 2024-09-17:
Just fyi, we just found and fixed one issue exactly like yours (stopped working after firmware update), and it was that signal "20" was partly shorted to the nearby ground plane:
[image]

20 is the leftmost signal, and it can happen that one scrapes the via there a bit too much, accidentally exposing the ground plane. Then when soldering the QSB down, the 2 can connect.
To know this, one can measure point 20 towards ground, and it should not read low ohms / fully connnected.

rdasilva84, 2024-09-17:
The strangest thing is that as shown in the video, the technician tested my Xstation PCB on another Playstation, model SCPH-100x and the symptom was the same. But in any case, the shipment was made today and as soon as I receive it I will let you know and we will start debugging. I have no experience with electronics or soldering, so I ask for your patience. I will do my best to.

ramapcsx2, 2024-09-17:
True. Forgot about that one :p

rdasilva84, 2024-09-18:
Isn't it related to my console being Japanese? The technician just sent me a message saying that he had just installed an Xstation in an American PlayStation and everything went well. When he took the PCB from the Xstation and put it in another Japanese console, the same problem occurred as mine. He has over 35 years of experience and is upset because he has installed several Xstations and has only seen this problem now in Japanese consoles =/

He has a YouTube channel where he has several videos installing Xstation and he is very curious to know what the solution will be for my case.

ramapcsx2, 2024-09-18:
The console region doesn't matter at all with xStation. It will run fine on all of them.

rdasilva84, 2024-09-24:
Hello.
I sent you a message on discord