x-station/xstation-issues #383: Issues after installation on PU-18

SodaPopin, 2024-09-30:
Hello. I tried to install my Xstation on my PS1. However, after the Sony logo, the screen will go to the memory card screen. Here's somethings of note:
-The blue light is on steadily but the yellow light for the SD card would blink at certain points. I hold the button that lets me run the disc while the cover's up when I tested this.
-The audio is gone for some reason
-I've reformatted my SD card a few times now. I placed the 2 necessary files in 00xstation folder.
-The SD card is 256 GB
-On the part with the QSB, I tested the points with a multimeter based on the guide here https://web.archive.org/web/20230221224926/https://mmmonkey.co.uk/xstation-sony-playstation-install-notes-and-pinout/. As far as I'm concerned, it passed.
-If I turn on the system with the cover up, it will go to the memory card screen faster (or rather normally) as compared to closing it down.
-Said PS1 used to have a modchip before installation but I removed it
-A config text file appeared once I reinserted my SD card to my PC

ramapcsx2, 2024-09-30:
Hey,
audio gone is your indicator. Did you accidentally lift the wrong chip's pins? Please check it's the lowermost chip (CD DSP), and not the SPU :)

SodaPopin, 2024-09-30:
> Hey, audio gone is your indicator. Did you accidentally lift the wrong chip's pins? Please check it's the lowermost chip (CD DSP), and not the SPU :)
I'll try to take a picture. Hopefully it's just a quick fix. Is it safe to assume then that the QSB is safe?

ramapcsx2, 2024-09-30:
For now, what you describe is just the generic "not quite working" case.
Xstation behaves normal, generating the config file and blinking as it should.
You get a picture also, meaning the APLL is working and correct.
If any one of the QSB signals is not working right, you get your result. It's sadly very unspecific.

SodaPopin, 2024-09-30:
> For now, what you describe is just the generic "not quite working" case. Xstation behaves normal, generating the config file and blinking as it should. You get a picture also, meaning the APLL is [...]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Pardon the size, not too familiar with github posting. By the way, is it safe to cut or trim the lifted pins?

ramapcsx2, 2024-09-30:
So.. In all fairness, let me recommend you send this to a modder for rescue .. :p

ramapcsx2, 2024-10-01:
That silence.. :p
I'm sorry, should have said a bit more.
The soldering work has several issues, making it unlikely that there is just one mistake.
The DSP are with the lifted pins looks like a battlefield! :p
To clean up the DSP, use isoprop on some qtips, carefully remove all the excess flux.
You can leave the lifted pins as they are. They won't move or get in the way. They're fine.

It seems you may have uses silver based solder, which is very difficult to work with. Regular old lead based would have been easier.
Also the QSB should be flush to the mainboard when soldering, so the solder doesn't have to bridge large distances..

Ideally you send this to a modder, because i strongly suspect there will be some problems.

SodaPopin, 2024-10-01:
> That silence.. :p I'm sorry, should have said a bit more. The soldering work has several issues, making it unlikely that there is just one mistake. The DSP area with the lifted pins looks like a battlefield! :p To clean up the DSP, use isoprop on some qtips, carefully remove all the excess flux. You can leave the lifted pins as they are. They won't move or get in the way. They're fine.
> 
> It seems you may have used silver based solder, which is very difficult to work with. Regular old lead based would have been easier. Also the QSB should be flush to the mainboard when soldering, so the solder doesn't have to bridge large distances..
> 
> Ideally you send this to a modder, because i strongly suspect there will be some problems.

Ah, pardon the silent treatment. Had to contact a modder. Based on our conversation, he suggested to recheck the DSP. Aside from your suggestion, I think I might have to ensure the pins next to the lifted ones. For the QSB, he also suggested to connect the number 17 from the pinout guide (The one from the link I sent) to either the QCB itself (The one were the ribbon cable is connected) or the part were the QSB is soldered to the PCB (The one on the right/the line based on the pinout guide). 
Regardless, thanks for the infos.

SodaPopin, 2024-10-01:
Will try to update

ramapcsx2, 2024-10-01:
Note, the information is still available here:
https://github.com/x-station/xstation-issues?tab=readme-ov-file#pu-18

Signal 17 is PSX level Reset, which sometimes is an issue, but not right now in your case.
Yours.. all the DSP area needs cleaning it up and ensuring no bridges / solder balls etc are there.

Then on the QSB, some solder points need to reflow, but this is hard, due to the silver based solder.
[image]
^ possibly a short here?

[image]
^ short?

[image]
^ trace ripped?

SodaPopin, 2024-10-01:
Aight, I'll try to modify around that part as well.