x-station/xstation-issues #385: Can start with custom BIOS?

mhmetalheart, 2024-10-26:
I own 2 original X-stations. I love them. And I usually play some Translated games which the translate team always put the characters into the BIOS. So, those game cannot start correctly on original console.
I played MiSTer FPGA and that will load the custom BIOS if put the BIOS together with the CUE+BIN and renamed as cd_bios.rom. So that when the game started it will automatically load that BIOS.
So, are there any chance to make the X-station do the same thing like MiSTer? E.X.: I put the XXX.rom together with the CUE+BIN, IN SAME folder, when start the game it will use that XXX.rom to cheat the console BIOS. So that I can play the Translated game.

Just a request. If really really couldn't, please ignore.

ramapcsx2, 2024-10-26:
Hey,
I don't really understand what's going on here, 2 things:
- the patched games require a custom PSX bios to work?
- the games currently fail on xStation, presumably because the BIOS is wrong?

I can definitely say that replacing the BIOS is not an xStation function. Like, not at all doable :p

mhmetalheart, 2024-10-27:
Yes, the patched game need a custom bios. If no, the game can run, but translated characters won't show correctly. I understand. No chance to run at all. :(

closed 2024-10-27 by mhmetalheart

ramapcsx2, 2024-10-27:
Go back to the patch authors: There is no way they cannot find some bytes in the exe to patch their font into :p