x-station/xstation-issues #386: Vampire Hunter D crashes during the credits sequence

NormanFoxLee, 2024-11-04:
SCPH-1001
Firmware: 2.0.2 Feb 4 2023

1. Play Vampire Hunter D (USA) - SLUS-01138, Version 1.00, 458 MB
2. Finish the game, wait till the credits.

Actual results:
Right after "The motion picture directed by Yoshiaki Kawajiri" the game crashes and you see the next screen:
[image]

After ~30 seconds console restarts into xStation menu.

Expected result:
The credits sequence plays w/o any issues.

ramapcsx2, 2024-11-04:
Hey,
wow, that is some crass behaviour. When I have a game crashing, it tends to black screen at the worst, but never outright reset the console on a system level.

To see this, I suppose I need a Memory Card save?

ramapcsx2, 2024-11-04:
Idk, I seem to vaguely remember having seen this before, but yea, all I can do here is check the game's first minutes or so, and they run fine, with nothing in particular in the logs. It seems to be a very normal game.

NormanFoxLee, 2024-11-07:
> Idk, I seem to vaguely remember having seen this before, but yea, all I can do here is check the game's first minutes or so, and they run fine, with nothing in particular in the logs. It seems to be a very normal game.

Hmm. I hope I did not delete the save or can restore it. If it is still here, on weekend I will try to extract the save via softmodded PS2. Also will try to reproduce on other PS1 model.

There were completely no problems at all in the game till the credits after the last boss fight.

NormanFoxLee, 2024-11-10:
Yeah, the save was deleted. Not possible to restore. I will replay the game on easy difficulty and hope that it will reproduce.
It will take some time, of course.