x-station/xstation-issues #387: Harry Potter and the Sorcerer's Stone freezes after the final credits when loading collected cards

NormanFoxLee, 2024-11-10:
No save files, sadly as:

1. Play Harry Potter and the Sorcerer's Stone (SLUS-01415), Version: 1.00, 298 MB
2. Beat the final boss
3. Save the game after the credits

Actual result:
After the save game loads album with cards but does not open it and not responding to the controls, ambient sound still plays.

Expected result:
Game suppose to open the album on its own.

In addition:
If you press reset button and load from the save - it will load the album and open it w/o any problem.

[image]

ramapcsx2, 2024-11-10:
I really need your savegames and instructions as well ><

NormanFoxLee, 2024-11-10:
As the last save rewrote the save that can reproduce this - I need to replay the whole game again.
Btw, I am trying to play and finish every PS1 game. So far out of 55 games only 2 had such problems)

ramapcsx2, 2024-11-10:
I understand it might not always be possible to have a good savegame, but the one thing I really can't do is play these games to completion :p
Additionally, I've long been working on the next firmware, which should be ready pretty soon, and which refines the core in various ways... it could mean these problems are gone by now.
Still would be good to have test cases, where it at least used to show problems.

NormanFoxLee, 2025-01-11:
Ok, looks like I understand what was going on. If you play any game long enough (like 6h) at some point of time it can freeze like this during loading some new environment in the game.

ramapcsx2, 2025-01-14:
Hey, sorry for the delay.
Testing such things is impossible (to get any usable information out of anyway).
Instead, have you tried the new beta update yet? It's pinned on Discord:
https://discord.gg/RRyxdjdH2r

The beta is going well, and you can always go back to another firmware, so it's worth a try.

NormanFoxLee, 2025-01-16:
Tomorrow I will have another long game session (~7h) - will use this 220 update.

NormanFoxLee, 2025-01-20:
So far no reproduction on beta version. Just sometimes some interruptions in sound on result screen (after the fight) in Grandia (same on release version).

ramapcsx2, 2025-01-23:
Hm, please try a different SD card. That sounds a lot like the card has slow times.
Any other card is fine. You only want to see if changing the card changes the problem (ideally fixes it), so you know you need a better one.