x-station/xstation-issues #388: Queuing discs and launching fails to work

DariotheHungry, 2024-11-11:
[image]

Hello! When I queue multiple discs and fast load or full load, I get the message I posted and then it resets to the beginning of the game list. 

Launching disc one works fine, but trying to queue and launch resets. 

Ive added and taken away plenty of games so I know the SD card isn't locked. Curious if anyone can help. (On latest firmware)

ramapcsx2, 2024-11-11:
Hey,
this would happen if somehow the diskqueue.txt file couldn't be written to / read from properly.
Best just try to delete that file, or better yet, the entire 00xstation folder (then recreate just the folder).

ramapcsx2, 2024-11-11:
I believe this has been addressed in all the socials and such now, right? But yea, the file or folder corrupted. Delete and have it create new fixes it.

DariotheHungry, 2024-11-12:
> Hey, this would happen if somehow the diskqueue.txt file couldn't be written to / read from properly. Best just try to delete that file, or better yet, the entire 00xstation folder (then recreate just the folder).

Thank you!