x-station/xstation-issues #390: The xstation not working

Vanez2, 2024-11-29:
I'm having an issue with the xstation. I can't get to the xstation menu. Green led only blinking few times when booting. I'm only getting the Memory Card/ Music Player screen. And some trouble with pointing when i`m trying to go to cd player in menu, PS dont recognize key presses when the lid is closed, if lid is closed no problem with key presses. 
Every recommended solution in Mini FAQ can`t solve this problem.

I alredy test soldering of the QSB, every point - good contact.
Also every pin on flex cable tested and its ok.

Here's photos of my solder work, and video with troble in menu.

https://youtu.be/MXOOzSujOvw?si=esLrm_OXm1eyJ5QM
[image]
[image]

ramapcsx2, 2024-11-29:
Hi,
you have the generic problem that means "one or more signals are not connected". The behaviour that you can't press buttons in the BIOS menu is normal. This occurs when the system tries to read a disk. To enter the CD player, have the Lid open, enter the player first, then close the lid. (Please do that to check if the system can play a silent audio CD fine, which will confirm some signals working or not.)

Your soldering looks great, so I'll assume you did a good job there, however.. I assume the system had an old modchip, that was taped down and removed by some force. Check out this:
[image]
That does not look good :p

ramapcsx2, 2024-11-29:
I can spot more just in your sample pics:
[image]

The most likely problem is that motherboard traces might be ripped, or that the QSB soldering connects the target signal with a bare nearby trace. Using a multimeter, you could check whether nearby spots are shorted together. If they are, that will be a tricky repair..

ramapcsx2, 2024-12-03:
Anything new? :p

Vanez2, 2024-12-04:
Yes modchip was been removed carefully, but it was installed not qute good.
Now i`ll trying to chek every trace on motherbord...
[image]

Vanez2, 2024-12-04:
motherboard traces is ok, despite looks like, the QSB soldering connects are good.
every point is used by xstation have a good connection with motherboard,

ramapcsx2, 2024-12-04:
Please do the CD Player test next. This helps narrow in on the problem.
```
Start the console with the CD lid opened, enter the music player, close the lid and it should find a CD to play.
If the CD plays, then the timer should advance the seconds normally, and you should be able to fast forward or reverse with the shoulder buttons.
```

Vanez2, 2024-12-15:
When lid is closed, ps don't recognise key presses, so i can't push play

ramapcsx2, 2024-12-16:
Enter the music player first, then close the lid.

Vanez2, 2024-12-19:
It's no matter, first going to music player or closing lid first, the same problem with input

ramapcsx2, 2024-12-20:
This means the console is trying to initialize the CD, but never succeeds. I still think this motherboard is bad, just from the damage seen in the picture.
I would recommend you try and get a different PSX, as even if you say you tested all the traces, I know how hard it is to actually ensure all is well.

Vanez2, 2025-01-23:
After many hours of testing, i found with oscilloscope bad singlal on  CXD1815Q , peplase this chip, but result not qute good...
Then i bought another PS1(japanese,without any modifications)  and installed xstation on it, now works well.

closed 2025-01-23 by Vanez2

ramapcsx2, 2025-01-23:
The  CXD1815Q is the data decoder, and is essential for the system to work. 
It is also known to randomly die, especially on the older PU-8 and earlier mainboards.
Sometimes it goes bad on PU-18 as well, which is your case.
Since you replaced the chip, and it still didn't work, it might also be the RAM that is attached to it, though I've only seen that once.

Perhaps what you saw looked a bit like this?
https://i.imgur.com/09S9Tts.mp4

Chip damage like this makes it really hard to help people, as it's impossible to debug without oscilloscope :/