x-station/xstation-issues #391: xStation vs Emulator Skipping

mabka33, 2024-12-02:
Hi, guys.  I use a 2012 era desktop and emulator with a mechanical HD for PS1 games.  I did some testing and the average read speeds of my hard drive are ~100 MB/s.  On a SanDisk Cruzer I tested for fun, the average was over 180 MB/s and reading about SD cards, they seem to have much higher numbers than my mechanical HD.

So, in short, I wanted to switch to the xStation, but the skipping issues I'm reading about concern me.  I'm curious as to why if both the xStation and emulators read the same files and the HD in my computer has slower read speeds than many SD cards, why emulators never seem to skip?  Aren't they subject to the issue of reading too slowly from the HD?

ramapcsx2, 2024-12-03:
Hey ho,
so in short: When you read about problems, what you don't see is the majority of people for which it works just fine. It's only those where their cards cause trouble that seek help.

The problem with such cards is that their internal firmware does not like the access pattern that xStation needs (which is brief small data size requests, continuously only ever asking for some 2kB).
Even those cards might produce an impressive 80MB/s or such reading in a fast PC, but this is because the bigger machines have lots of RAM and will request data in much larger chunks.

All in all, if you want to make sure, get a Sandisk Extreme series card (red/gold), or the new Samsung "Pro" (in blue) cards.
They are certified good with their controller firmware.

mabka33, 2024-12-03:
Thank you, sir.

closed 2024-12-03 by mabka33