x-station/xstation-issues #392: [HELP] Alternative reset point for late PU-8

vsilvalopes, 2024-12-23:
Hi !

After moving my QSB from my original PU-8 PS1, to a new one, I can't get the Xstation to reset the console or do the full boot when starting a Game. The problem is, ever since I've moved the Xstation (and the QSB) to another system, I've probably damaged the reset pin on the QSB.

The Fast Load work just fine, and I can play all my games with no issue.

So I would like to know, if on the PU-8 motherboard is an alternative reset point, like the one described here :
https://github.com/x-station/xstation-issues?tab=readme-ov-file

Also, from that alternate point, in wich pin of the Xstation QSB I have to solder it ?

Thanks in advance!

ramapcsx2, 2024-12-26:
[image]
This will be your RST point on the QSB, and judging by the location of it, perhaps you only have to reflow it to make it work...
But if you need an alternative, the connector where the power supply plugs in has that reset signal, on the most "northward" pin:
[image]
(It reaches to the mainboard underside as well.)

vsilvalopes, 2025-01-20:
Hi @ramapcsx2 !

Thanks for the quick repply (and sorry I'm my delay to answer back).
Your tip on the reflow was killer! 

I did it and everything is working fine! I was unsure if this was the right pin, but with your help reflowing it was a breze!
Good to know that we have other reset points on the board! 

Thanks again for the help and sorry the delay to report back!

ramapcsx2, 2025-01-20:
Happy to help :)

closed 2025-01-20 by ramapcsx2